olikraus/u8g2 #1353: Bug on UpdateDisplayArea on U8G2_ST7920_128X64_F_SW_SPI

franzSusemichel, 2020-12-06:
I have an issue when trying the example code for UpdateDisplayArea with U8G2_ST7920_128X64_F_SW_SPI.  The letters look torn. It is easy to reproduce for me when I'm displaying only a single letter. The Pixels are on correct Y-Position but shifted on x-Position.

olikraus, 2020-12-06:
Thanks for creating the issue here.
However the bad news is, it is not supported for ST7920 as mentioned in Note 1 here: https://github.com/olikraus/u8g2/wiki/u8g2reference#updatedisplayarea

franzSusemichel, 2020-12-06:
Thx for your feedback. Haven't seen the notice. Do you think it will be supported in future?

olikraus, 2020-12-06:
Probably not, the mentioned controller have strange memory architectures. It should be possible to avoid this function. Maybe you can describe your problem do that we can find a different solution.

franzSusemichel, 2020-12-20:
Hi, 

I thought I have to use UpdateDisplayArea but it turns out that display is fast enough to updated at once. Thanks for your great support. 

Kind regards

closed 2020-12-20 by franzSusemichel