olikraus/u8g2 #1354: Quick and dirty solution from ST75256 -> ST75161 for WO128128A2-TFH

JBleckmann, 2020-12-06:
To everybody who may help
I made a quick and dirty solution by modify u8x8_d_st75256_wo256x128_init_seq to init 75161.
Unfortunately image is shifted to right side. 

[image]

I failed to find solution in init sequence:
`static const uint8_t u8x8_d_st75256_wo256x128_init_seq[] = {          //modified for ST75161
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */

  U8X8_C( 0x030 ),				// Extension Command 1
  U8X8_C( 0x06E ),			
  
  U8X8_C( 0x031 ),			    // Extension Command 2
  U8X8_CA( 0x0d7, 0x09f),	    // Set auto-read instruction, Disable Auto Read
  
  U8X8_CA( 0x0e0 ,0x000),      // Enable OTP Read
  U8X8_DLY(10),
  
  U8X8_C( 0x0e3 ),              // OTP Read
  U8X8_DLY(20),
  U8X8_C(0xe1),          // OTP Control Out
  
  U8X8_C(0x30),          // Extension Command 1
  U8X8_C(0x94),          // Sleep Out
  U8X8_C(0xae),          // Display OFF  
  U8X8_DLY(50),
    
  U8X8_CA(0x20,0x0b ),          //Power Control VB, VR,VF All ON
  
  U8X8_CAA(0x81, 0x0d,0x04 ),   // Set Vop = 14V
  
  U8X8_C(0x31),            // Extension Command 2
  U8X8_C(0x20),           // Set Gray Scale Level
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x1f),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x1f),
  U8X8_A(0x1f),
  U8X8_A(0x1f),
  U8X8_A(0x00),
  U8X8_A(0x00),

  U8X8_CAAA(0x32, 0x00, 0x01, 0x02),  // Analog Circuit SetBooster, Efficiency = Level 1, Bias=1/12

  U8X8_CA(0x51, 0xfb),          // Booster Level x10

  U8X8_CAAAA(0xf0, 0x02, 0x08, 0x0f, 0x18),  // frame rate setting

  U8X8_C(0x30),                 // Extension Command 2
  U8X8_CA( 0x0f0, 0x010 ),		/* monochrome mode  = 0x010*/

  U8X8_CAAA(0xca, 0x00, 0x87, 0x00),     /* display control, 3 args follow  */
  /* 0x00: no clock division, 0x04: devide clock */
  /* 1/136 duty value from the DS example code */
  /* Line cycles in a frame */

  U8X8_CA(0xbc, 0x03),      /*Column and page direction invers */

  U8X8_C(0xa6),       //Normal Display
  
  U8X8_C(0x31),       //Extension Command 2
  U8X8_C(0x40),       //Internal Power Supply

  U8X8_C( 0x030 ),				/* select 00 commands */
  U8X8_C( 0x077 ),				/* Enable ICON RAM */
  U8X8_CAA(0x15, 0x00, 0x7f),		/* col range */
  U8X8_C( 0x076 ),				/* Disable ICON RAM */
  U8X8_CAA(0x75, 0x00, 0x10),		/* row range */
  
  U8X8_C(0x30),
  U8X8_CAA(0x15, 0x00, 0xff),        // Column Address Setting SEG0 -> SEG159
  U8X8_CAA(0x75, 0x00, 0x10),        // Row Address Setting, Page address

  U8X8_C(0x30),      
  U8X8_CAA(0x15, 0x00, 0x87),         // Column Address Setting 135
  
  U8X8_C(0xaf),                 // Display ON
  
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};
`
Thanks 
Joachim

olikraus, 2020-12-06:
I think you need to set this value to 0:
https://github.com/olikraus/u8g2/blob/1b656cbd5d61d5a907a0f9f32e3b48fd5e2fc3cf/csrc/u8x8_d_st75256.c#L390

JBleckmann, 2020-12-06:
Thanks for fast response but
 /* default_x_offset = */ 0,	/* must be 0, because this is checked also for normal mode */
has no effect. Still shifted

olikraus, 2020-12-06:
It should have an effect. Maybe you did modify the wrong file in the wrong directory. Or Arduino IDE requires a restart.

JBleckmann, 2020-12-06:
Yes. Should have an effect. Thanks for hints, I have same thougts.
I need to dive deeper in ST75161-G2A specification about ICON RAM. I am not sure if example code from Winstar engineer  is correct.

Write_cmd(0x30);
Write_cmd(0x77);       // Enable ICON RAM
Write_cmd(0x15);        // Column Address Setting
Write_data(0x00);       // SEG0 -> SEG159 start address
Write_data(0x7f);       // end address 128
Write_cmd(0x76);        //Disable ICON RAM

Write_cmd(0x75);        // Row Address Setting
Write_data(0x00);       // COM0 -> COM159 start page address
Write_data(0x10);       // end page address

Write_cmd(0x30);
Write_cmd(0x15);        // Column Address Setting
Write_data(0x00);       // SEG0 -> SEG159
Write_data(0xff);
Write_cmd(0x75);        // Row Address Setting
Write_data(0x00);       // Page address
Write_data(0x10);
Write_cmd(0x30);        
Write_cmd(0x15);         // Column Address Setting
Write_data(0x00);
Write_data(0x87);         // 135
Write_cmd(0xaf);          // Display ON
}

JBleckmann, 2020-12-07:
I need more time to find root failure. I close the issue meanwhile.

JBleckmann, 2020-12-08:
I found my failure. As character were flipped I try to solve always by scan direction command 
>   U8X8_CA(0xbc, 0x06),      /*Column and page direction invers */
and u8g2 flipmode. 

But failure was data format.
Correct is 
<  U8X8_CA(0xbc, 0x00),      /*Column and page direction  */
<   U8X8_C(0x0C),                //data format 0C LSB top
[image]

Feel free to add ST75161 for WO256128A-TFH# to your library. 
/*=============================================*/
/* WO256X128, https://github.com/olikraus/u8g2/issues/891  */

static const u8x8_display_info_t u8x8_st75256_wo256x128_display_info =
{                                     // Modified for 75161
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
  
  /* post_chip_enable_wait_ns = */ 20,
  /* pre_chip_disable_wait_ns = */ 20,
  /* reset_pulse_width_ms = */ 5, 	
  /* post_reset_wait_ms = */ 5, 		/**/
  /* sda_setup_time_ns = */ 20,		/* */
  /* sck_pulse_width_ns = */ 40,	/*  */
  /* sck_clock_hz = */ 4000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,	/* 400KHz */
  /* data_setup_time_ns = */ 15,
  /* write_pulse_width_ns = */ 70,	
  /* tile_width = */ 32,
  /* tile_hight = */ 16,
  /* default_x_offset = */ 0,	/* must be 0, because this is checked also for normal mode */
  /* flipmode_x_offset = */ 0,		/* used as y offset */
  /* pixel_width = */ 128,
  /* pixel_height = */ 128
};

static const uint8_t u8x8_d_st75256_wo256x128_init_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */

  U8X8_C( 0x030 ),				// Extension Command 1
  U8X8_C( 0x06E ),			
  
  U8X8_C( 0x031 ),			    // Extension Command 2
  U8X8_CA( 0x0d7, 0x09f),	    // Set auto-read instruction, Disable Auto Read
  
  U8X8_CA( 0x0e0 ,0x000),      // Enable OTP Read
  U8X8_DLY(10),
  
  U8X8_C( 0x0e3 ),              // OTP Read
  U8X8_DLY(20),
  U8X8_C(0xe1),          // OTP Control Out
  
  U8X8_C(0x30),          // Extension Command 1
  U8X8_C(0x94),          // Sleep Out
  U8X8_C(0xae),          // Display OFF  
  U8X8_DLY(50),
    
  U8X8_CA(0x20,0x0b ),          //Power Control VB, VR,VF All ON
  
  U8X8_CAA(0x81, 0x0d,0x04 ),   // Set Vop = 14V
  
  U8X8_C(0x31),            // Extension Command 2
  U8X8_C(0x20),           // Set Gray Scale Level
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x1f),
  U8X8_A(0x00),
  U8X8_A(0x00),
  U8X8_A(0x1f),
  U8X8_A(0x1f),
  U8X8_A(0x1f),
  U8X8_A(0x00),
  U8X8_A(0x00),

  U8X8_CAAA(0x32, 0x00, 0x01, 0x02),  // Analog Circuit SetBooster, Efficiency = Level 1, Bias=1/12

  U8X8_CA(0x51, 0xfb),          // Booster Level x10

  U8X8_CAAAA(0xf0, 0x02, 0x08, 0x0f, 0x18),  // frame rate setting

  U8X8_C(0x30),                 // Extension Command 2
  U8X8_CA( 0x0f0, 0x010 ),		/* monochrome mode  = 0x010*/

  U8X8_CAAA(0xca, 0x00, 0x87, 0x00),     /* display control, 3 args follow  */
  /* 0x00: no clock division, 0x04: divide clock */
  /* 1/136 duty value from the DS example code */
  /* Line cycles in a frame */

  U8X8_CA(0xbc, 0x00),      /*Column and page direction  */
  U8X8_C(0x0C),       //data format 0C LSB top

  U8X8_C(0xa6),       //Normal Display
  
  U8X8_C(0x31),       //Extension Command 2
  U8X8_C(0x40),       //Internal Power Supply

  U8X8_C(0x30),
  U8X8_C( 0x077 ),				/* Enable ICON RAM */
  U8X8_CAA(0x15, 0x00, 0x9F),		/* col range */
  U8X8_C( 0x76 ),				/* Disable ICON RAM */
  U8X8_CAA(0x75, 0x00, 0x10),		/* row range */
  
  U8X8_CAA(0x15, 0x00, 0x9F),      // Column Address Setting SEG0 -> SEG159
  U8X8_CAA(0x75, 0x00, 0x10),       // Row Address Setting,  16 Page address

//  U8X8_C(0x30),      
//  U8X8_CAA(0x15, 0x00, 0x7f),         // Column Address Setting 135
  
  U8X8_C(0xaf),                 // Display ON
  
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

closed 2020-12-08 by JBleckmann