olikraus/u8g2 #1357: Issue with 64128K, Arduino MEGA and u8g2lib

fernandojr-88, 2020-12-10:
Dear 

I hope somebody can help me with this

I had been trying to run the U8g2lib on a Arduino MEGA and a 64128K display with ST7565 controller (https://cdn.displaytech-us.com/sites/default/files/display-data-sheet/64128K%20series.pdf)

I`m using a level shifter for interface the MEGA and the display, also I try to use Adafruit library for ST7565 and it worked normally.

But when i try to use the U8g2lib, i just see a clean screen.

The construcor i`m using is this:

U8G2_ST7565_64128N_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);

I was trying to run the HelloWorld example

Any help would be much apreciatted

olikraus, 2020-12-25:
Did you crosscheck whether the pin numbers fit?
There might be also a contrast issue. You could run the contrast-test example and check whether something will be visible.

https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/ContrastTest/ContrastTest.ino

fernandojr-88, 2020-12-27:
Hi Olikraus

First of all thank you for your attention.

I build and uploaded the contrasttest.ino, but is still the same, nothing shows up on the display.

The constructor I'm using is the below one:

U8G2_ST7565_64128N_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);

I had changed the pins to match the pins of the Adafruit library, and still with the same wires connection Adafruit libraries seens to work and U8g2lib not. 

I used your library on a project with another project and i love it, so i look forward to use it again.

Best Regards

olikraus, 2020-12-27:
If the adafruit lib works then also u8g2 lib will work. Maybe you can post the adafriut setup here.

fernandojr-88, 2020-12-28:
Below is the code I`m using from Adafruit, i did not change anything from the example.

Only on the U8g2 example I made changes, changing the pins assingments to match the ones on the Adafruit library

The adafruit code:

`include "ST7565.h"

int ledPin =  13;    // LED connected to digital pin 13

// the LCD backlight is connected up to a pin so you can turn it on & off
#define BACKLIGHT_LED 4

// pin 9 - Serial data out (SID)
// pin 8 - Serial clock out (SCLK)
// pin 7 - Data/Command select (RS or A0)
// pin 6 - LCD reset (RST)
// pin 5 - LCD chip select (CS)
ST7565 glcd(9, 8, 7, 6, 5);

#define LOGO16_GLCD_HEIGHT 16 
#define LOGO16_GLCD_WIDTH  16 

// a bitmap of a 16x16 fruit icon
const static unsigned char __attribute__ ((progmem)) logo16_glcd_bmp[]={
0x30, 0xf0, 0xf0, 0xf0, 0xf0, 0x30, 0xf8, 0xbe, 0x9f, 0xff, 0xf8, 0xc0, 0xc0, 0xc0, 0x80, 0x00, 
0x20, 0x3c, 0x3f, 0x3f, 0x1f, 0x19, 0x1f, 0x7b, 0xfb, 0xfe, 0xfe, 0x07, 0x07, 0x07, 0x03, 0x00, };

// The setup() method runs once, when the sketch starts
void setup()   {                
  Serial.begin(9600);

#ifdef __AVR__
  Serial.print(freeRam());
#endif

  // turn on backlight
  pinMode(BACKLIGHT_LED, OUTPUT);
  digitalWrite(BACKLIGHT_LED, LOW);

  // initialize and set the contrast to 0x18
  glcd.begin(0x18);

  glcd.display(); // show splashscreen
  delay(2000);
  glcd.clear();

  // draw a single pixel
  glcd.setpixel(10, 10, BLACK);
  glcd.display();        // show the changes to the buffer
  delay(2000);
  glcd.clear();

  // draw many lines
  testdrawline();
  glcd.display();       // show the lines
  delay(2000);
  glcd.clear();

  // draw rectangles
  testdrawrect();
  glcd.display();
  delay(2000);
  glcd.clear();

  // draw multiple rectangles
  testfillrect();
  glcd.display();
  delay(2000);
  glcd.clear();
[... 193 lines cut ...]
  //pinMode(18, OUTPUT);
  //digitalWrite(18, 1);	
  
  u8g2.begin();  
}

void loop(void) {
  uint8_t i;
  i = 255;
  do
  {
    testContrast(255-i);
    delay(15);
    i-=5;
  } while( i != 0 );
  i = 255;
  delay(100);
  do
  {
    testContrast(i);
    delay(15);
    i-=5;
  } while( i != 0 );
  delay(100);
} `

olikraus, 2020-12-28:
All looks correct. Did you test any other ST7565 constructors?

fernandojr-88, 2020-12-29:
Hi 

I try the contrast example with all the  below constructors, only the ones for  ST7565 and with SW_SPI, none of they seen to work
```
//U8G2_ST7565_64128N_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 8, / data=/ 9, / cs=/ 5, / dc=/ 7, / reset=*/ 6);
//U8G2_ST7565_EA_DOGM128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);
//U8G2_ST7565_64128N_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);
//U8G2_ST7565_EA_DOGM132_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ U8X8_PIN_NONE);  // DOGM132 Shield
//U8G2_ST7565_ZOLEN_128X64_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);
//U8G2_ST7565_LM6059_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);    // Adafruit ST7565 GLCD

/// THE KS0713  also did not compile
//U8G2_ST7565_KS0713_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);    // KS0713 controller

//U8G2_ST7565_LX12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);
//U8G2_ST7565_ERC12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);

////The ERC12854 did not compile
//U8G2_ST7565_ERC12864_ALT_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6); // contrast improved version for ERC12864

//U8G2_ST7565_NHD_C12832_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);
//U8G2_ST7565_NHD_C12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);
U8G2_ST7565_JLX12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 8, /* data=*/ 9, /* cs=*/ 5, /* dc=*/ 7, /* reset=*/ 6);
```
Also i notice that the led "L" is always ON when i run the sketch with any of those constructors (is this a normal behaviour on this example?)

Best Regards

olikraus, 2021-05-09:
Not sure how I can help here. Hope it's ok to close...

closed 2021-05-09 by olikraus