olikraus/u8g2 #1358: Add support for UC1611S controller, 256x128 display

spoland99, 2020-12-11:
Hello, 

I've been trying to use your library with my display: 
manufacturer part number: CI064-4073-06 (Intelligent Display Solutions)
Controller: UC1611S
RS components number: 124-7546

Its a pretty nice display, and seems to be quite high resolution for its size, but I'm struggling to find support for it. I'd like to display monochrome bitmaps (.xbm) on it using 4W_SW_SPI. I've tried your .xbm example using the following constructor:
U8G2_UC1611_EA_DOGXL240_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 

Its quite buggy and the .xbm's aren't being displayed correctly - I think due to the resolution of the 2 displays being slightly different:
-> EA_DOGXL240: 240 x 128
-> CI064-4073-06: 256 x 128

Is there any way to modify the library to add support for the: CI064-4073-06?

Many thanks, 

Sam :)

spoland99, 2020-12-12:
Update:

I tried modifying the library myself :

1) I modified u8x8_d_uc11.h by adding the code below. I named the display ids4073, and copy pasted the code for the ea_dogxl240. I just changed the tile_width and pixel_width and replaced 240x128 by 256x128: 
`
/*================================================*/
/* IDS4073 */

static const uint8_t u8x8_d_uc1611_ids4073_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x02f),            			/* internal pump control */
  U8X8_CA(0x0f1, 0x07f),			/* set COM end */
  U8X8_CA(0x0f2, 0x000),		/* display line start */
  U8X8_CA(0x0f3, 127),		/* display line end */
  U8X8_C(0x0a3),            			/* line rate */
  U8X8_CA(0x081, 0x08f),		/* set contrast */
  
  //U8X8_C(0x0a9),            			/* display enable */

  U8X8_C(0x0d1),            			/* display pattern */  
  U8X8_C(0x089),            			/* auto increment */
  U8X8_CA(0x0c0, 0x004),            	/* LCD Mapping */
  U8X8_C(0x000),		                /* column low nibble */
  U8X8_C(0x010),		                /* column high nibble */  
  U8X8_C(0x060),		                /* page adr low */
  U8X8_C(0x070),		                /* page adr high */
  
  
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

static const u8x8_display_info_t u8x8_uc1611_256x128_display_info =
{
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
  
  /* post_chip_enable_wait_ns = */ 10,	/* uc1611 datasheet, page 60, actually 0 */
  /* pre_chip_disable_wait_ns = */ 10,	/* uc1611 datasheet, page 60, actually 0 */
  /* reset_pulse_width_ms = */ 1, 
  /* post_reset_wait_ms = */ 10, 	/* uc1611 datasheet, page 67 */
  /* sda_setup_time_ns = */ 10,		/* uc1611 datasheet, page 64, actually 0 */
  /* sck_pulse_width_ns = */ 60,	/* half of cycle time  */
  /* sck_clock_hz = */ 8000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 30,	/* uc1611 datasheet, page 60 */
  /* write_pulse_width_ns = */ 80,	/* uc1611 datasheet, page 60 */
  /* tile_width = */ 32,		/* width of 30*8=240 pixel */
  /* tile_hight = */ 16,
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 256,
  /* pixel_height = */ 128
};

/* UC1611s 256x128 display */
uint8_t u8x8_d_uc1611_ids4073(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  /* call common procedure first and handle messages there */
  if ( u8x8_d_uc1611_common(u8x8, msg, arg_int, arg_ptr) == 0 )
  {
    /* msg not handled, then try here */
    switch(msg)
    {
      case U8X8_MSG_DISPLAY_SETUP_MEMORY:
	u8x8_d_helper_display_setup_memory(u8x8, &u8x8_uc1611_256x128_display_info);
	break;
      case U8X8_MSG_DISPLAY_INIT:
	u8x8_d_helper_display_init(u8x8);
	u8x8_cad_SendSequence(u8x8, u8x8_d_uc1611_ids4073_init_seq);
	break;
      case U8X8_MSG_DISPLAY_SET_POWER_SAVE:
	if ( arg_int == 0 )
	  u8x8_cad_SendSequence(u8x8, u8x8_d_uc1611s_powersave0_seq);
	else
	  u8x8_cad_SendSequence(u8x8, u8x8_d_uc1611s_powersave1_seq);
	break;
      case U8X8_MSG_DISPLAY_SET_FLIP_MODE:
	if ( arg_int == 0 )
	{
	  u8x8_cad_SendSequence(u8x8, u8x8_d_uc1611s_flip0_seq);
	  u8x8->x_offset = u8x8->display_info->default_x_offset;
	}
	else
	{
	  u8x8_cad_SendSequence(u8x8, u8x8_d_uc1611s_flip1_seq);
	  u8x8->x_offset = u8x8->display_info->flipmode_x_offset;
	}	
	break;
      default:
	return 0;		/* msg unknown */
    }
  }
  return 1;
}
`
2) I added the device name to u8x8.h: 
         `uint8_t u8x8_d_uc1611_ids4073(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);`

3) I updated the codebuild.c file by adding the following: 
```

{
      "uc1611", 	32, 	16, 	"u8g2_ll_hvline_vertical_top_lsb", "u8x8_cad_001", "", COM_4WSPI|COM_3WSPI|COM_6800|COM_8080,
    "", /* is_generate_u8g2_class= */ 1,
    {
      { "ids4073" }, //256x128
      { NULL }
    }
  },  
  {
    "uc1611", 	32, 	16, 	"u8g2_ll_hvline_vertical_top_lsb", "u8x8_cad_uc16xx_i2c", "i2c", COM_I2C,
    "", /* is_generate_u8g2_class= */ 1,
    {
      { "ids4073" }, //256x128
      { NULL }
    }
  }, 
```

4) I compiled and executed codebuild.c and replaced the files in the arduino library with the new ones codebuild.c had generated. 

5) When I compiled in the Arduino IDE using the U8x8lib.h library and U8X8_UC1611_IDS4073_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); , the display works perfectly and I can use the full width of the screen.

6) When I tried using the U8g2lib.h and U8G2_UC1611_IDS4073_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); , the code compiles but the display is blank. 

Is there something I've missed that is preventing the U8g2lib.h from working with my display? 

Many thanks, 

Sam

olikraus, 2020-12-25:
> Is there something I've missed that is preventing the U8g2lib.h from working with my display?

the uc1611 common procedure will not work for the 256x128 display, it needs to be rewritten.

olikraus, 2020-12-25:
I have added the suggested constructor. Can you test whether this works?
I have created beta  2.28.9

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2020-12-27:
Closing this for now. Nevertheless any feedback whether this works is highly welcome.

closed 2020-12-27 by olikraus

spoland99, 2020-12-27:
Hi Oli, 

Thanks very much for doing this! 

I'm back home from Uni (and my hardware) at the moment but when I get back I'll test this. 
Have a happy Christmas!!
Thanks, 

Sam :)