olikraus/u8g2 #1360: U8g2 freeze the screen SH1106 I2C on Arduino

diego2938, 2020-12-15:
Hi, 
I am reporting a problem with the library U8g2 with SH1106 I2C on Arduino (MEGA 2560).

The screen is this one: "amazon.it/AZDelivery-Display-retroilluminato-Raspberry-gratuito/dp/B078J78R45"
I am using the constructor line: U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

With the "GraphicsTest" examples of the library (as well as my own code) after a while the screen freezes, this is true for all the mode of operation: Full buffer, page buffer and u8x8.
(I have tested successfully the screen with the "u8glib" and the "wonho-maker/Adafruit_SH1106" libraries).

Since this problem is common with other users (https://forum.arduino.cc/index.php?topic=690214.0), I was wondering if you are aware of the problem and if you are planning to solve it. 

Thanks in advance
Best regards
Diego

olikraus, 2020-12-21:
Thanks for the information. I never found a bug in u8g2 which causes such freeze (which does not mean that such a bug doesn't exist), instead the most often issue had been:
- low memory 
- config data corrupted in OLED 

One of the important questions is this: Does you uC still work (maybe do some led blink in parallel) or is the uC frozen.
If the uC still works, try to issue another u8g2.begin(): Will this unfreeze the OLED?

andrew-newell, 2020-12-22:
I believe I might be running into this same issue. I've purchased a 0.96" OLED from amazon and after a short period of time (I think it might be # of sendBuffer() calls, not time - but I'm not sure), my screen freezes and nothing recovers it except removing the power. (I've tried `.clear()`, `.clearDisplay()`, `.clearBuffer()`, `.begin()`,`.initDisplay()`,)  I'd be more than happy to coordinate with you @olikraus if you'd like to debug through me to confirm if what I'm experiencing is the same as others.

After some more testing, even disconnecting the display itself, it never recovers (screen was finally cleared but would never activate again) - It looks like something gets stuck in the esp8266 and persists across resets until power loss.

diego2938, 2020-12-22:
Dear olikraus, thank you for the answer. I do have the OLED with me and I'll be able to re-test it after the Christmas holidays. I'll be back with the answers.

Dear andrew-newell, I am currently using the OLED 0.96'' with the u8g2 library without any problem. Did your screen also freeze with the "GraphicsTest" examples?

Best regards.
Diego

olikraus, 2020-12-22:
There had been issues with esp system in the past. Maybe try a page buffer example and avoid sendBuffer command.

andrew-newell, 2020-12-22:
> Dear olikraus, thank you for the answer. I do not have the OLED with me and I'll be able to re-test it after the Christmas holidays. I'll be back with the answers.
> 
> Dear andrew-newell, I am currently using the OLED 0.96'' with the u8g2 library without any problem. Did your screen also freeze with the "GraphicsTest" examples?
> 
> Best regards.
> Diego

@diego2938 Yes, I was unable to use the GraphicsTest examples with either of the two working constructors I have that function properly with text.

> There had been issues with esp system in the past. Maybe try a page buffer example and avoid sendBuffer command.

@olikraus I'll see what I can figure out with page buffers and report back.

andrew-newell, 2020-12-22:
@diego2938  Okay for whatever with the nodemcu v3 esp8266 board I'm able to get GraphicsTest to work correctly now with my I2C 0.96 monitor. Here's the constructor I'm using:
`U8G2_SSD1306_128X64_NONAME_F_HW_I2C gfx(U8G2_R0, /* reset=*/U8X8_PIN_NONE);`

I'll see what I can do about my code to work with the page buffer stuff.

olikraus, 2020-12-25:
Maybe this is a watchdog / yield issue. Whill wdtDisable() help?

diego2938, 2020-12-29:
> Thanks for the information. I never found a bug in u8g2 which causes such freeze (which does not mean that such a bug doesn't exist), instead the most often issue had been:
>
> * low memory
> * config [...]

Hi olikraus,

I had the chance to test the OLED: I can confirm you that both the uC and the OLED freeze.
I did the test combining the blink example with your GraphicsTest (u8x8), just to make reproducible things. 

Should I provide other "symptoms" that could help in solving the problem? 

Best regards.
Diego

olikraus, 2020-12-29:
ok, I think we should separate things a little bit
On ESP systems, the problem might be caused by the watchdog issue
On AVRMega systems I never heard about such issues and I actually do not know how to support here.

andrew-newell, 2021-01-05:
> ok, I think we should separate things a little bit
> On ESP systems, the problem might be caused by the watchdog issue
> On AVRMega systems I never heard about such issues and I actually do not know how to support here.

Sorry for not responding sooner, the holidays got the better of me. After moving to the `U8G2_SSD1306_128X64_VCOMH0_F_HW_I2C` I didn't get any further errors - though I have had to put the project to the side for the moment and only tested for a handful of minutes at a time. I should be getting back to the OLED part of my project and at that time I'll conduct some stability testing for nonstop operation. If I run into anything I will update back here.

Cheers!

andrew-newell, 2021-01-12:
Just FYI, as it turns out, some devices don't play nice on the I2C bus such as the AHT10 sensor I was using initially. I believe freezes caused by that sensor as well as a bad configuration with my OLED screen caused a surplus of crashes/hangs with this project. I haven't gotten back to this project to reproduce the issues, but I'm reasonably certain that the AHT10 sensor was contributing to my frustrations. 

https://wiki.liutyi.info/display/ARDUINO/AHT10?preview=/30507639/30507715/Aosong_AHT10_en_draft_0c.pdf

[image]

I accidentally came across this info from a post (https://www.raspberrypi.org/forums/viewtopic.php?t=265734) while researching sensors for my project O_o. Thought it might be relevant to other users in the future to double check what is on the I2C bus for compatibility.

closed 2021-05-09 by olikraus