olikraus/u8g2 #1362: SSD1306 adressing mode not setup correctly. SSD1306 works by accident only?

TobleMiner, 2020-12-19:
Hi,

I'm currently writing a SSD1306 emulator. While testing compatibility different libraries I found that u8g2 setups ssd1306 based OLED displays incorrectly.

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x64_noname.c#L53 (and many more places like it) claims to setup the display in page addressing mode. However, addressing mode 0x00 is horizontal addressing mode, not page addressing mode. See https://cdn-shop.adafruit.com/datasheets/SSD1306.pdf page 30.

Most amazingly the library does still work since u8g2_send_buffer sends all tiles (which are really just the ssd1306 gddram pages) in order as full rows. Thus while the set low/high column  and set start page commands are ignored, it just overflows into the next page and by accident continues drawing in just the right spot.

Am I missing something, or is it really only working by accident?

TobleMiner, 2020-12-19:
Please see #1363 for my suggested fix

olikraus, 2020-12-21:
You are correct  regarding the comment. The comments in the .c are wrong. U8g2 will use the horizontal mode.
From my understanding the difference between the horizontal and the page addressing mode is just the page increment, which is missing in the page addressing mode.

> Thus while the set low/high column and set start page commands are ignored

Not true, because u8x8 works as expected. The u8x8 subsystem of u8g2 is based on the correct behavior of the set col / row commands. 

> it just overflows into the next page and by accident continues drawing in just the right spot.

It could overflow in u8x8 with a large string, but probably will never overflow in u8g2. From that perspective it probably would not matter whether u8g2/u8x8 will use horizontal or page mode.

> Am I missing something, or is it really only working by accident?

The only difference between the two modes is the page increment, which is not used in u8g2.
From my perspective everything works as intended.

olikraus, 2020-12-21:
I have created test code for this. As expected, the only difference is the page increment. The code draws two tiles at the right edge of the display. The first tile still fits at the right border, the second wraps to the left edge. It will depend on the mode, whether the second tile appears in the same row (page mode) or next row (horizontal mode).

I do not see any accidents here :-)

``` C++
// two 6x6 pixel frames centered within two 8x8 tiles
uint8_t tile[16] = {
  0x00,
  0x7e,
  0x42,
  0x42,
  0x42,
  0x42,
  0x7e,
  0x00,

  0x00,
  0x7e,
  0x42,
  0x42,
  0x42,
  0x42,
  0x7e,
  0x00
};
...
  u8x8.setFont(u8x8_font_amstrad_cpc_extended_r);
  u8x8.drawTile(15,0, 2, tile);
  u8x8.drawString(0, 2, "This is line y=2");
```
mode 0 (horizontal mode), page is incremented, u8g2 mode for SSD1306
[image]

mode 2 (page mode), page is NOT incremented
[image]

olikraus, 2020-12-21:
Arduino example is here: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8x8/DrawTile/DrawTile.ino

olikraus, 2020-12-25:
updated several comments

closed 2020-12-25 by olikraus