olikraus/u8g2 #1364: JLX25664 on ESP32S SPI BUS (HSPI)

LightningTerry, 2020-12-23:
I use

 - [ESP32S2 WROVER](https://www.espressif.com/sites/default/files/documentation/esp32-s2-wrover_esp32-s2-wrover-i_datasheet_en.pdf)
 - [JLX25664](https://www.mouser.ch/datasheet/2/291/NHD_3_12_25664UCW2-1902065.pdf)

And I am trying to connect these two together. But somehow nothing is displayed, neither I receive any Signals from the GPIO13(MOSI).

<details open>
<summary>I tried using some Demo Codes Like this:</summary>
<br>
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>

#define HSPI_MOSI_PIN 13
#define HSPI_MISO_PIN 12
#define HSPI_SCK_PIN  14
#define HSPI_CS_PIN   15

U8G2_ST75256_JLX25664_F_4W_HW_SPI u8g2(U8G2_R0, /* CS=*/ HSPI_CS_PIN, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();          // clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");  // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  delay(100);
}
</details>

But It does not work.

I even tried things, suggested in here #1331

v1993, 2020-12-23:
Please put your code inside ```. Includes are so broken by parsing them as html tags.

olikraus, 2020-12-23:
Did you try the sw constructor? At least to check whether the wiring is ok.

LightningTerry, 2020-12-28:
@olikraus do you mean using this instead?
```cpp
#define SPI_MOSI_PIN 13
#define SPI_MISO_PIN 12
#define SPI_SCK_PIN  14
#define SPI_CS_PIN   15
#define SPI_DC_PIN   10

//U8G2_ST75256_JLX25664_F_4W_HW_SPI u8g2(U8G2_R0, /* CS=*/ SPI_CS_PIN, /* reset=*/ U8X8_PIN_NONE);
U8G2_ST75256_JLX25664_F_4W_SW_SPI u8g2(U8G2_R0, /*SCK=*/SPI_SCK_PIN , /*MOSI=*/SPI_MOSI_PIN, /*CS=*/SPI_CS_PIN, /*DC=*/SPI_DC_PIN , U8X8_PIN_NONE);
```

Should there be any difference? It still does not show any error message.

olikraus, 2020-12-28:
I also strongly suggest to connect the reset pin.

LightningTerry, 2020-12-28:
So as i was connecting the reset pin, i noticed that my pinout I was using, did not match with the description in the datasheet, so i changed some other pins aswell  (_I updated the Code in the Issue Description_). But even with that it still doesn't show anything on the Display.

olikraus, 2020-12-28:
In your initial post you have referenced a display with SSD1322 controller. Your constructor is for a display of the same size but with a ST75256 controller. Obviously this will not work.

If you own the display described in the reference under JLX25664, then you should use one of the following constructors:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1322-nhd_256x64

Also note, that you need to enable U8g2 16 bit mode (unless you use the latest u8g2 release, which will enable 16 bit mode automatically for your board)

LightningTerry, 2020-12-29:
Thank you, with this change I finally could Display something for the first time :)
But somehow i had a white square on the top right corner of the Display. And after I did a reset i couldn't display anything anymore, with the same Code.

LightningTerry, 2020-12-29:
I am not sure if this also helps, but i just noticed, that the code I use, the "Hello World!" should be displayed 10 pixels far from the left border. But that one time that it displayed the text was directly at the coordinates 0,0.
[image]

olikraus, 2020-12-29:
I think the box is caused by 8 bit mode. Did you change to 16 bit mode (see FAQ)?

LightningTerry, 2020-12-29:
yes, I uncommented the Line ```#define U8G2_16BIT```

olikraus, 2020-12-29:
> yes, I uncommented the Line #define U8G2_16BIT

so everything works now?

LightningTerry, 2020-12-29:
Sorry, i meant that i already had the line uncommented. But i can't even check anymore if it works, because somehow since i pressed the reset button it doesn't show anything at all on the display.

olikraus, 2020-12-29:
oh

LightningTerry, 2020-12-30:
Update: If I change the delay to 500ms, I have almost everytime something on the Display, sometimes it doen't work, but i guess it's a hardware problem. Besides this, it also doesn't show anything if I use the drawStr on the wrong Coordinates or use Values between 10..15...20.. and so on. And the Display somehow doesn't use its full size, it cuts the bottom of the text already at a pointer(10,20).

[image]

olikraus, 2020-12-30:
> Besides this, it also doesn't show anything if I use the drawStr on the wrong Coordinates or use Values between 10..15...20.. and so on.

You have not sent any code, so it is difficult to say what is wrong here, but from the picture I would say, you have used the "_1_" constructor but instead you should use the "_F_" constructor.

LightningTerry, 2020-12-30:
Is "U8G2_ST75256_JLX25664_**_F_**_4W_SW_SPI" not the "_F_" constructor?
If it's not, what would it be, or what should I change?

olikraus, 2020-12-30:
hmm, so the code in your initial post will exactly show the above picture?
I have a request: Can you execute the u8x8 hello world? How will it look like?
You probably need to uncomment this line and add replace your personal pinnumbers:
https://github.com/olikraus/u8g2/blob/3c6460a73f7f310c665cef0af1d2bac49bf6c655/sys/arduino/u8x8/HelloWorld/HelloWorld.ino#L100

By default the "hello world" will be printed in the second line:
https://github.com/olikraus/u8g2/blob/3c6460a73f7f310c665cef0af1d2bac49bf6c655/sys/arduino/u8x8/HelloWorld/HelloWorld.ino#L261

Will it appear?
What will happen if you change the y position? Will hello world always appear in the correct line?

LightningTerry, 2020-12-31:
Okay, if I Use it on Position 0,1 then it looks like this:
[image]
But with this Code i can move the text up to position 20,6
[image]
I thought, that the white Box appears, because it isn't able to display the "!". After removing it and just displaying "Hello World" the box disappears, but instead the "H" in the beginning behaves weird.
[image]

olikraus, 2020-12-31:
Very strange, I never saw such a behavior before.
Let's try to see, whether there is a timing issue:
https://github.com/olikraus/u8g2/blob/3c6460a73f7f310c665cef0af1d2bac49bf6c655/csrc/u8x8_d_ssd1322.c#L247-L248

Please locate u8x8_d_ssd1322.c on you local harddisk (should be in the arduino library folder) and double/increase the above mentioned values. Especially the post cs delay might have cause the problems.
Will this help?

LightningTerry, 2020-12-31:
Changing the values also changes the pattern of the animated pixels so it really might be something with the Chip Select. But even after playing around with the values I couldn't get rid of the error. 
And its not always the "H", sometimes, it's also the "d" at the end, it switches randomly when I reset. Maybe that helps.

olikraus, 2020-12-31:
Can you choose a different pin for the CS signal? Maybe also change wiring for the CS signal completely.

LightningTerry, 2020-12-31:
I tried that, but sadly didn't change anything.

olikraus, 2021-05-09:
oh, long time pending topic here. 
Did you solve the problem?

closed 2022-03-04 by olikraus