olikraus/u8g2 #1368: sh1106 128x64 purchased from ebay

kgamble01, 2020-12-26:
Thank you for your files, we tried many sources but the only one that works are your u8g2 files.    However, there is still one small problem.
The top 7 or 8 rows are not showing correctly - photo attached

if i move the printing of these lines further down the screen they print correctly.  
the only error i can see (but do not understand) is " avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description" which shows at end of succuessful compliation.

I have shown your inverse print u8g8 library in the photo - 
[image]
, but error is there for other text or graphic if printed in that area.
The rest of the screen prints perfectly.  We have 2 of these displays and both do the same, so i think it is software not hardware.

olikraus, 2020-12-26:
Actually the complete screen is wrong. For example the "g" appears more like "3".
Which constructor did you use?

Can you use the  U8G2_SSD1306_128X64_ALT0_1_4W_HW_SPI constructor and reply back with a picture of this constructor. Expectation would be, that the text is correct (but instead there will be a two pixel width garbage column)
If this the U8G2_SSD1306_128X64_ALT0_1_4W_HW_SPI works, we could implement a U8G2_SH1106_128X64_ALT0_1_4W_HW_SPI constructor.

kgamble01, 2020-12-26:
WOW so soon!  Just doing it now!  Thanks😊

kgamble01, 2020-12-26:
I should have said it is I2c with no reset – the photos are with the following constructor:

U8X8_SSD1306_128X64_ALT0_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);              // same as the NONAME variant, but may solve the "every 2nd line skipped" problem

 

 

Thanks very much

Karen

olikraus, 2020-12-26:
>  I should have said it is I2c with no reset – the photos are with the following constructor:
U8X8_SSD1306_128X64_ALT0_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);              // same as the NONAME variant, but may solve the "every 2nd line skipped" problem

oh, in this case you should use the U8X8_SSD1306_128X64_NONAME_HW_I2C constructor
As written in the comment above, the ALT0 will skip every second line. This is exactly what you can see in the picture.
The ALT0 constructor is for very rare SSD1306 OLEDs with a special row interleaving architecture.

kgamble01, 2020-12-26:
Hello,  still a problem at the top and has small bar at far right which is probably the difference between 1306 and 1106

 

Really, really appreciate your help as we have been struggling for days.

Thanks

olikraus, 2020-12-26:
> Hello,  still a problem at the top and has small bar at far right which is probably the difference between 1306 and 1106

Yes, sounds like a SH1106 display then.

kgamble01, 2020-12-26:
Hello,  yes it is 1106.  It seems strange to me that it is only the top 8 lines that are wrong, below that the display seems correct.  We can use it as it is because we need only to display a counter in a large font (maximum value will always be less than 1000) and one line at the bottom to show which one of  two readings it is showing.  
Because we dont have the top 8 lines, the font at the bottom has to be very very small.  
We are using "arduino nano every" to drive it.  
Thanks

olikraus, 2020-12-26:
>  It seems strange to me that it is only the top 8 lines that are wrong

On the image you posted it seems that the complete picture is wrong.

I assume, it you would use "U8G2_SH1106_128X64_NONAME_1_HW_I2C", then all problems will be gone.

kgamble01, 2020-12-26:
Hello

  I cannot find that exact file. The closest I can come is …..U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

 

I think I am looking for U8g2  SH1106  128x64 noname  1   HW I2c

*	U8g2 (is the library)
*	SH1106 is the chip
*	Noname is no-name😊
*	1 is buffer size
*	HW I2C  - two wire
*	No reset – only has 4 pins

 

The one above has buffer problems  and will not run properly at all  “Problems with full buffer mode and skipped clear” and I think this is because of the “F” which indicates there is full buffer, and I don’t think on the Nano Every this can be the case?

kgamble01, 2020-12-26:
VICTORY!!!!!  We changed over to the other display (which we have used before) but now the second display is working perfectly.

 

U8X8_SH1106_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE); is now working.

Thanks so much for your help.

Really really appreciate it.

Thanks   and have a wonderful New Year

closed 2021-05-09 by olikraus