olikraus/u8g2 #1369: Wiring esp8266 D1 mini / JLX256160G

Bruno999B, 2020-12-26:
Hello, would there be someone to help me, to wire / operate a JLX256160G SPI 3.3V 4 wire lcd, on an ESP8266 D1 mini? I have been trying for several hours and I can only turn on the backlight ... 
Thanks in advance for any help

[image]

olikraus, 2020-12-26:
1. Read the datasheet
2. Create a mapping & wiring table between your display and your Arduino board

In detail:

1. You need to check the datasheet to find, clock, data, A0 and CS lines. In chapter  7.4.1 of the data sheet (file:///tmp/mozilla_kraus0/ST75256_v1.6b.pdf) you find:

D0=SCL=Clock 
D1, D2, D3 = Data
D4 .. D7 = 3.3V 
ERD = 3.3V
RWR = 3.3V
A0 = A0
CSB = CS

These datasheet names have to be compared with labels on your display PCB:

| Datasheet Label | PCB Label | U8g2 Name |
|----------|-----------|---------|
| D0=SCL | D0 | Clock |
| D1,D2,D3=Data | D1, D2, D3 | Data |
| ERD | E | - |
| RWR | R/W | - |
| A0 | RS | DC |
| CSB | CS | CS |

2. Add your wiring: Connect your board to the above mentioned pins and add the pin numbers of your board in another column; 

| Datasheet Label | PCB Label | U8g2 Name | Arduino Board |
|----------|-----------|---------|--------|
| D0=SCL | D0 | Clock | ? |
| D1,D2,D3=Data | D1, D2, D3 | Data | ? |
| ERD | E | - | 3.3V |
| RWR | R/W | - | 3.3V |
| A0 | RS | DC |  ? |
| CSB | CS | CS | ? |

The pin numbers given in the above table finally have to be provided as arguments to the U8g2 constructor.

See also: https://github.com/olikraus/u8g2/wiki/setup_tutorial

Bruno999B, 2020-12-26:
thanks to your help I was able to make my screen work, I had found the datasheet but even with it, I had not managed to make it work. I would have liked to test the different operating speeds using the "FPS" example but this one does not work.

on ESP8266 with u8g2 library this problem appears on several examples

`
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
`
thank you again for giving me all this information, happy end of year celebration !

olikraus, 2020-12-27:
If you have issues with ESP systems: It could be caused by the ESP watchdog which needs to be disabled for u8g2.

Bruno999B, 2020-12-28:
I tried with "ESP.wdtDisable();" in setup, but it does not work. there is still the problem of the hardware watch dogs. 
```
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
```
do I have to feed the dog periodically?
Thanks for your help

olikraus, 2020-12-28:
My knowledge on esp systems is almost zero. 🙄

Bruno999B, 2020-12-28:
ok thanks, no problem, I can already use the screen now! Thank you !
to help others ...:
Screen JLX256160G-680 ---> https://fr.aliexpress.com/item/1876223568.html
ESP8266 d1 mini

`U8G2_ST75256_JLX256160_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ D8, /* dc=*/ D3, /* reset=*/ D0);   // Enable U8g2 16 bit
[image]

`

N° | PCB | U8G2 | ESP8266
-- | -- | -- | --
1 | NC | NC | NC
2 | NC | NC | NC
3 | NC | NC | NC
4 | NC | NC | NC
5 | LEDA | +3,3V | +3,3V
6 | VSS | GND | GND
7 | VDD | +3,3V | +3,3V
8 | RS | DC | D3
9 | RESET | RESET | D0
10 | CS | CS | D8
11 | D7 | +3,3V | +3,3V
12 | D6 | +3,3V | +3,3V
13 | D5 | +3,3V | +3,3V
14 | D4 | +3,3V | +3,3V
15 | D3 | DATA | D7
16 | D2 | DATA | D7
17 | D1 | DATA | D7
18 | D0 | CLOCK | D5
19 | E | +3,3V | +3,3V
20 | R/W | +3,3V | +3,3V

Works at SPI 10 Mhz
3D printing case, STL available if anyone wants

olikraus, 2020-12-28:
👍

closed 2021-05-09 by olikraus

dynek, 2022-03-24:
> 3D printing case, STL available if anyone wants

I'm interested by the STL 🙏