olikraus/u8g2 #1372: SSD1322 graphical issues

wanthalf, 2020-12-27:
I have strange graphical issue with SSD1322 using U8G2_SSD1322_NHD_256X64_F_3W_HW_SPI u8g2(U8G2_R0, A4) on STM32F103. I have not checked the example codes, but the issues seem to only concern some graphical operations - possibly lines and not bitmaps? See attached image from the IconMenu example. The same appears in the GraphicsTest where rectangles and circles are drawn, but not where text is rendered (except of rotated text).

[image]

Possibly, you will have an immediate idea what the problem may be. Otherwise, I can do more tests of particular operations, of course. I also have more pieces of the same display that I have not tested yet, but to me it does not look like a problem of the display.

wanthalf, 2020-12-27:
I will add a few more shots from the problematic operations within the GraphicsTest.

[image]
[image]
[image]
[image]

wanthalf, 2020-12-27:
Oh, The issues only concern the full buffer mode, not the page buffer! Everything seems to work well in the page_buffer GraphicsTest.

wanthalf, 2020-12-28:
Finally one obvious bug discovered: u8g2.getDisplayWidth() returns just 240 instead of 256.

I noticed the page buffer GraphicsTest does not use the full width of the display. The last 16(?) pixels (on x-axis) are "dead". When rotated upside down (U8G2_R2), it is the "first" 16 pixels. Actually, the "dead area" is the area, where the graphical overruns/underruns appear in full buffer mode. Unfortunately, I cannot find more than the fact that the u8x8_display_info_t u8x8_ssd1322_256x64_display_info structure defines the pixel width correctly as 256.

AnHardt, 2020-12-28:
[FAQ](https://github.com/olikraus/u8g2/blob/master/doc/faq.txt)

> : How to activate 16 Bit mode?
A: Search for the line
//#define U8G2_16BIT
in "u8g2.h". Uncomment this line:
#define U8G2_16BIT
The file "u8g2.h" is located in "/libraries/U8g2_Arduino/src/clib" inside your default
sketch folder.

wanthalf, 2020-12-28:
Funny, I just read this particular Q&A few days ago and wondered what is the "16 bit mode" and when should it be used.  :-D

Well, problem solved. Thanks. I suppose I will need to change this every time I use another of my displays again. Would be more practical to put it into the code somewhere if possible, but never mind.

This is a wonderful library. It could save a lot of time for both users and developers if the documentation were comprehensible and not spread across various documents, FAQs and other files. This was probably my third unnecessary report, I am sorry for that.

olikraus, 2020-12-28:
I have just released u8g2 2.28.10 which will activate 16 bit mode by default on stm32 boards.

wanthalf, 2020-12-28:
I see that the "16 bit mode" is necessary because of the display width >= 256, not because of the controller. So, I suppose I will need it also when the display is used with an ATmega2560, right? What is turned to "16 bit" - all the communication with the display? So, does it have any negative consequences (slowing the communication by half)?

wanthalf, 2020-12-28:
I also see the documentation has improved a lot! I would suggest adding a more explicit link to the FAQ for people to consult the wiring and particular display specifics somewhere in the initial pages (the introduction?). And also add the information to the FAQ for these displays, that the "enable" signal should be connected to /WR and that the /RD pin should be permanently connected to Vcc, as I found somewhere else.

olikraus, 2020-12-28:
> And also add the information to the FAQ for these displays, that the "enable" signal should be connected to /WR and that the /RD pin should be permanently connected to Vcc, as I found somewhere else.

Well, using u8g2 still means you need to read the display datasheet. I can not describe details for all available displays ;-)
Which signal is mapped to which pin and what to do with unused pins, still is described in the display datasheets.

> What is turned to "16 bit"

Only the data type which holds the pixel coordinates in u8g2 are changed from 8 bit to 16 bit.
This means:

 * U8x8 is unaffected: U8x8 will use a "tile" coordinate system (1 tile = 8x8 pixel). As a result u8x8 can work with display dimensions of up to 2040x2040
 * In U8g2 8 bit mode all pixel related calculations are done in 8 bit, which is faster and produces smaller code on 8 bit architectures like ATMega AVR. This will however restrict the display size to 255x255 (which is actually clipped at 240x240)
 * In U8g2 16 bit mode the pixel related calculations are done in 16 bit. For 32 bit systems like ARM, there will be not much a difference between 16 bit mode and 8 bit mode, this is also why 16 bit mode is now default on ARMs. The maximum display dimension is then 65000x65000, however, because u8g2 just calls u8x8, the actual limit in 16 bit mode is 2040x2040 pixel.

wanthalf, 2020-12-28:
Thanks for clarification. I have not studied the internals, but this gives already much more hints about the low level functionality hidden in U8g2. It seems the average user should still know more of the hidden low-level principles.

While the datasheets are quite clear concerning signals such as "chip select", "write", "read" and "data/command", it is still not clear what a generic term such as "enable" means in terms of U8g2 (unless you understand the internals), and how it differs from plain "chip select". Also, a beginner cannot be sure whether the controller will actually ever need to "read" from the controller, or not.
I understand the difficulty of supporting so many different displays and controllers by a single library and I admire all the effort, but it can be expected that it will attract naive hobbyists like me, who can find some spare time to find out how to use such a high-level library, but not to study all the details of the single controllers and their internal principles (otherwise, I could just write my own library myself). And yes, I also know the frustration of users who are too impatient (or just lazy) to read ALL the documentation I have written for them, so I am sorry for offering such a poor stupid-users feedback myself :-)

Please, copy-paste your above mentioned information to the documentation or FAQ as well, it is really very helpful! I have also found the information about wiring the /WR and /RD signals here in the "issues", but not in the documentation.

olikraus, 2020-12-28:
Thanks for your clarification and please excause if my answers had been too short and did appear too grumpy. Probably 50% of all the issues (we reached issue 1372) here are more or less wiring questions. 

I will reopen the issue and assign myself the task to update the docs.

Nevertheless, some description is her: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#wiring
In U8g2 I have introduced the pin name "enable" for the 6800 and 8080 interface. It is often called W/R or E, but you can never be sure how this signal is called on the display module PCB or the display datasheet.

But truly, I should mention, that any RD signal should be connected to "H". However, whether it is called RD again depends on the datasheet and differs between displays.

On the other side, if you would use a controller like the ST7920 (which is one of the most worse controllers), everything again is wrong, and "E" again has a total different meaning and RD has inverted logic in parallel mode:
[image]
and https://github.com/olikraus/u8g2/wiki/gallery#26-nov-2016-st7920-128x64-lcd-in-8080-parallel-mode

So finally, there is no general role and users have to read the manuals to some extend.
I mean, there is also a risk that I confuse users instead of providing advice.

Best is to create a mapping table. Post the same here as an issue and we can discuss how the display should be connected.

There is a nice video on this topic (not from be, but regarding my v1 of u8glib): https://www.youtube.com/watch?v=4CD8ERaylmY&list=PL89YvYOmjljsv2qFkWxm2lyrm1RHxyAQh&index=3

Or as another example, see a parallel discussion here: https://github.com/olikraus/u8g2/issues/1369

olikraus, 2021-05-09:
updated the docs & faq...

closed 2021-05-09 by olikraus