olikraus/u8g2 #1374: 2 software emulated ssd1306

matheusfillipe, 2020-12-29:
Is it possible to have two software emulated i2c? I am using this constructor:
`U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 21, /* data=*/ 20);`

The hardware ones didn't work and my display isn't detected by the i2c scanner. I got two of them i would like to plug on my mega board. Would that be possible?

olikraus, 2020-12-29:
Yes, should be possible. Please also check the u8g2 FAQ.

matheusfillipe, 2020-12-29:
Oh thanks... I just finally understood how this works. Do you mind telling a bit more of how SW works(going off topic here but just got curious :P)? Why this software emulation works for devices like mine that aren't detected as an i2c device?

closed 2020-12-29 by matheusfillipe

olikraus, 2020-12-29:
The software I2C device implementation the I2C protocol on bit level without using the I2C hardware available on your uC. 
The I2C protocol includes a feedback information from the client, which will tell whether the client has accepted the information or not. This feedback is also used by I2C scanner to find any available client. 
The feedback information is also used to slow down the communication in cases where the client needs more time to process incoming data. 
In cases of display controllers, the transfer information will be processed fast enough and will be accepted all the time. To simplify and speed up the software I2C emulation: U8g2 does not check for this feedback (because it is anyways always true).

As a consequence: If the display does not send the feedback at all, then u8g2 sw I2C will still work while hardware I2C whill wait for the feedback, which will never arrive.