olikraus/u8g2 #1378: ESP32 + SSD1306 OLED freezes after the very first output

dma7, 2021-01-03:
I'm running the latest version 2.28.10 of u8g2 lib on a ESP-32S .
I have two I2C devices on my board: the display (128x64 OLED) and the other one is the BME280 sensor. IMO, the wiring of SDA/SCL is according to the recommendations, i.e. in chain with a pull-up resistor between the controller and the first device (display). BTW, the examples run perfectly fine without any modifications of the wiring.

When I place the following piece of code 

```
    u8g2.begin();
    u8g2.clearBuffer();        
    u8g2.setFont(u8g2_font_ncenB08_tr); 
    u8g2.drawStr(0,10,"Booting OK"); 
    u8g2.sendBuffer(); 

```

at the very beginning of the setup() method, it works fine showing the short message.
When I shift this snippet to the very end of setup() nothing happens anymore (i.e. blank screen)
The same is true (i.e. "blank screen"), when I place the code in the loop() method.

What happens in between? My sketch bases on 
https://github.com/adlerweb/ESP8266-BME280-Multi
which means that in starts an HTTP server and connects to a MQTT broker. Although the display stays empty all the other functionality keeps working (i.e. no crash or reboot)

I ported the original code to the ESP32 and I worked perfectly fine until I started taking the next step to integrate the display. 

Any ideas what could be wrong? 

Thanks,
Daniel

olikraus, 2021-01-04:
> pull-up resistor between the controller and the first device

Not sure what this means: There must be two pullup resistors from the uC pin to 3.3V 
Do the BME280 and SSD1306 share the same pin? If so, you need to ensure, that there is no conflict with between the BME280 lib and u8g2: This means, both libs must use the same I2C method (like HW I2C) and should use the same bus speed. See also the U8g2 FAQ: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt

dma7, 2021-01-04:
Thank you for the hint. I changed the ctor to 

`
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
`
and now everything works as expected. 

BTW, I think one resistor per wire (SCL and SDA) should be enough. See 
[https://www.mikrocontroller.net/topic/88806]()

Thanks, 
Daniel

closed 2021-01-04 by dma7