olikraus/u8g2 #1380: Winstar WG12864D with T6963C /CD pin inversion

vergo77, 2021-01-05:
First of all, Hello and many thanks for this amazing library!
I've connected my Arduino UNO with the WG12864D GLCD.
At the beginning nothig was displayed on screen, all the connections was correct, also on VSM program it doesn't worked.
After signals checking I've noticed the /CE signal was inverted, after placeing a NOT logic port on VSM it started to work,  then i've  connected a CD4069 on the real circuit and it started to work as well.
Finally the output /CD signal from the library seems to be inverterd, so my question is: Is there a way to invert by SW the /CD signal?

Thanks a lot and Happy new year to all!

vergo77, 2021-01-26:
I give an answer to my self and for who will have the same issue:
I found where invert the CS for my case:
file: **u8x8_d_t6963.c** 
lines: **382-383**
Action: I've swapped 0 and 1 togheter and now it works properly.
`  /* chip_enable_level = */ 0,

  /* chip_disable_level = */ 1,
`
I've also noticed that "chip_enable_level" and "chip_disable_level" are eve in other places of this file, but in thise cases 0 and 1 are like my edit.
Maybe this is an error on the source code, but I cannot say that.

olikraus, 2021-01-30:
Thanks for your input. CS somehow are standard to the controller. Looks like your board does some inversion here. Fixing the u8g2 at the mentioned location is probably the best.

closed 2021-03-07 by olikraus