olikraus/u8g2 #1383: GitHub 'too many request abuse' system triggered on fonts wiki page

TontonSancho, 2021-01-11:
The following wiki page contains too many individual embedded image references :
**[U8g2 Fonts](https://github.com/olikraus/u8g2/wiki/fntlistall)** page at address : https://github.com/olikraus/u8g2/wiki/fntlistall

Lot of PNG references can't load and you may occasionally trigger GitHub too many request abuse system, if you try to reload this page.

The **[Font Groups](https://github.com/olikraus/u8g2/wiki/fntgrp)** wiki page does not have this problem since it is organized in sub-categories.

May be the **[U8g2 Fonts](https://github.com/olikraus/u8g2/wiki/fntlistall)** page should be re-organized in sub-categories as well.

Regards.

Elijahg, 2021-02-28:
I get the same problem, and it disables the entirety of Github for a while! Also makes it impossible to view fonts past about 7 pixels high.

olikraus, 2021-03-07:
yes :-(

https://github.com/olikraus/u8g2/issues/1304

olikraus, 2021-07-09:
I have splitted the fntlistall page into four sub pages. Hope this helps.

closed 2021-07-09 by olikraus