olikraus/u8g2 #1391: Visual artifacts with 2 displays using UpdateDisplayArea

matheusfillipe, 2021-01-26:
I have a mega 2560 board connected to 2 ssd1322 which were initialized as:

``` 
U8G2_SSD1306_128X64_NONAME_F_SW_I2C screen1(U8G2_R0, /* clock=*/ SCREEN1[0], /* data=*/ SCREEN1[1]);
U8G2_SSD1306_128X64_NONAME_F_SW_I2C screen2(U8G2_R0, /* clock=*/ SCREEN2[0], /* data=*/ SCREEN2[1]);

``````
They are slow to update entirely and at some part of my code I want them to
fastly update one area on each one at the same time (alternating). The problem
is that part of what is on the buffer of one gets written on the other
generating those artifacts. I tried cleaning all buffers before writing but it
didn't help. I am writing to each of them as:

``` 

void write(){
     screen1->clearBuffer();
     screen2->clearBuffer(); //Thought it could help
     screen1->clear(); // Also doesn't help
     screen1->drawStr(10,20, "Some string here");
     screen1->updateDisplayArea(0, 2, 26, 4);
 }
`````

And on the second one is the same thing, just replace 1 by 2 there and 2 by 1.
I added all those clears thinking it could help but had no effect at all. When
I first call write like that it works fine, but as soon as write is called for
the other display then I get part of what should go to one on the other.

matheusfillipe, 2021-01-27:
My bad! doing clearbuffer totally works to solve this issue.

closed 2021-01-27 by matheusfillipe

JamesGKent, 2021-01-27:
U8G2 uses the same buffer internally in situations like this, so you have to clear, render one display then update it, then clear and render the other.
Also they are slow because you're using software I2C, hardware would be faster, but to use them both that way you'll need to look at changing the address of one of them, both in hardware (usually a pin to pull to ground or supply) and then software to match.