olikraus/u8g2 #1392: I2C Problem

ercanersoy, 2021-01-28:
Hello,

I have found on a I2C problem with using multiple I2C devices.

I have used SSD1306 OLED display with DS1307 RTC. I have used hardware I2C bus on Arduino Uno. When changes a page on the OLED display, date and time data corrupts in RTC.

I think u8g2 library does not pause I2C bus for other devices sharing bus with the OLED screen.

Thanks,
Ercan

JamesGKent, 2021-01-28:
SSD1306 display supports I2C at 400KHz, the DS1307 does not. if the RTC library is poorly written and does not check the bus frequency before using it as per arduino guidance then it will fail to work. you should be able to either set the clock to 100KHz yourself prior to using the RTC, or set the bus clock to 100KHz in u8g2

ercanersoy, 2021-01-30:
The solution works. Thanks a lot.

closed 2021-02-06 by ercanersoy

JPRIPKE, 2024-09-10:
Hi, can you show to me where do that change? either the rtc lib or the u8g2 lib.

olikraus, 2024-09-10:
See the notes here: https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock