olikraus/u8g2 #1395: u8x8 Missing some open iconic fonts

Seitekful, 2021-02-02:
The u8x8 library only supports a couple of open_iconic_xxx fontgroups. u8g2 supports all and the u8x8 library could benefit from more icons.

AntiSol, 2021-04-15:
Hi. This isn't exactly what you're asking for, but it's not very difficult to convert any font you like. I have had some success creating better fonts using bdfconv and font editors like gbdfed and fontforge.

I found the documentation a little unclear WRT generating u8x8 fonts but after some experimentation I found that the following command line works for me:

```
./bdfconv -b 3 -f 2 -th 2 -tv 2 -m '64-73' -n varname -o /path/to/output.c /path/to/fontfile.bdf
```

where -th and -tv set how many 8x8 characters wide/tall each glyph is. In this example I'm creating a set of 2x2 icons. For 4x4 you would use -th 4 -tv 4 instead.

Hope this helps! :)

olikraus, 2021-05-09:
Closing this...

closed 2021-05-09 by olikraus