olikraus/u8g2 #1402: u8g2 integration in with st7565_nhd_c12864 using c language simplicity studio

chandramauli-mpc, 2021-02-08:
I am trying to implement u8g2 drivers  for st7565_nhd_c12864 in c language using SPI . Please let me know the sample code . I tried base codes of examples but i can't see anything in my screen.

chandramauli-mpc, 2021-02-11:
@olikraus i am trying to implement on simplicity studio with EFM32 If anyone has example code please let me know. TIA.

olikraus, 2021-02-14:
Did you read this page: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform ?
It should give you all the required information.

3wire HW SPI is a little bit more complicated. If EFM32 support 9 bit SPI tokens, then I suggest to use this, otherwise you could follow the approach in the Arduino part:
https://github.com/olikraus/u8g2/blob/3c6460a73f7f310c665cef0af1d2bac49bf6c655/cppsrc/U8x8lib.cpp#L749

closed 2022-03-04 by olikraus