olikraus/u8g2 #1403: Library support for applying an offset to the screen area

broddo, 2021-02-09:
Hi - this isn't an issue really, just a question.

Say I have a 128x64 LCD but I only ever want to draw between pixels 4 and 120 horizontally and say 2 and 60 vertically (basically defining the screen as {4, 2, 120, 60} ). Then if I call, say u8g2_DrawBox(&lcd, 0, 0, 10, 10) it would automatically put the top left corner of the box at 4,2. Is there an easy way for me to set this up using the library or would I better off just doing this manually?

olikraus, 2021-02-14:
You can use https://github.com/olikraus/u8g2/wiki/u8g2reference#setclipwindow for the reduced display area, but there is no build-in method for the request xy translation.

broddo, 2021-02-15:
Setclipwindow() certainly helps with half of my problem. I had also spotted the user->x0 and user->y0 properties and thought I could set them myself but that doesn't have the desired effect. I'll just add my own translation layer then. Thank you!

closed 2021-02-15 by broddo