olikraus/u8g2 #1404: Won't display text

dxvictor, 2021-02-09:
u8x8.drawString(0,0,"Temp:");  prints ":" only.

 u8x8.setCursor(0, 0);    
 u8x8.println("Temp:");

prints ":" only

dxvictor, 2021-02-09:
#include <U8x8lib.h>
U8X8_SH1106_128X64_NONAME_HW_I2C u8x8(U8X8_PIN_NONE);
void setup() {
  u8x8.begin();
  u8x8.setFont(u8x8_font_7x14B_1x2_n); 
  u8x8.clear();
  u8x8.drawString(0,0,"TEST"); //Nothing displayed
  delay(5000);
  u8x8.setCursor(0, 0);
  u8x8.print("TEST");   //Also nothing displayed
  delay(5000);
  u8x8.drawString(0,0,"1234"); //Displays fine
  delay(5000);
  u8x8.setCursor(0, 0);
  u8x8.print("4321");   //Displays Fine
  u8x8.drawString(0,0,"?!-:"); //Displays only -:  
}
void loop() {}

AnHardt, 2021-02-10:
Font Name | Character Set
-- | --
u8g2_xxx_xe | Extended: Glyphs with unicode 32 to 701 are included in the font (v2.16.x will also include big ß)
u8g2_xxx_xf | Full: Glyphs with unicode 32 to 255 are included in the font
u8g2_xxx_xr | Restricted: Only characters from 32 to 127 are included
u8g2_xxx_xu | Uppercase: Numbers and uppercase letters
u8g2_xxx_xn | Numbers and some extra glyphs for date and time printing are included
u8g2_xxx_x_something | Special selection of glyphs. See the font picture for details.

dxvictor, 2021-02-10:
Thanks. I didn't understand how the fonts work.

closed 2021-02-10 by dxvictor