olikraus/u8g2 #1405: "defined(unix)" is defined by gcc, but not by g++ on the same system

oxygene, 2021-02-11:
I'm compiled u8g2 as C static lib using GCC and was linking it with a C++ program compiled by g++ on the very same machine (Ubuntu 20.04 with gcc 9.3.0). My program kept crashing, libASAN claiming that I got buffer overflows on very simple operations.

After some time I found out that `sizeof u8g2_t` was different in the C library and in the C++ code. The reason turned out to be a difference in `U8G2_16BIT` being set in `u8g2.h` when compiled in C, but not in C++:
```cpp
#if defined(unix) || defined(__arm__) || defined(__xtensa__) || defined(xtensa) || defined(__arc__) || defined(ESP8266) || defined(ESP_PLATFORM)
#define U8G2_16BIT
#endif
```

It took me some time to find out that I was using g++ with `-std=c++17` which apparently makes g++ not define `unix` any more. Hence the difference to gcc where I didn't specify any extra flags.

Maybe there is a more robust way to determine whether we are on a platform with large ints? On GCC, you could check `__SIZEOF_INT__` for example (see https://gcc.gnu.org/onlinedocs/cpp/Common-Predefined-Macros.html)

olikraus, 2021-03-14:
on my system I get this:
```
kraus@b117m:/data/kraus/git/scad/card_compare_machine$ gcc -dM -E - < /dev/null | fgrep unix
#define __unix__ 1
#define __unix 1
#define unix 1
kraus@b117m:/data/kraus/git/scad/card_compare_machine$  g++ -dM -E -x c++ - < /dev/null | fgrep unix
#define __unix__ 1
#define __unix 1
#define unix 1
```

Maybe additionally I should check for __unix__

oxygene, 2021-03-14:
> Maybe additionally I should check for `__unix__`

My testing also confirms that `g++` seems to always define `__unix__`, no matter if `-std=c++17` is specified or not. So this might do the trick. Thanks!

olikraus, 2021-04-20:
See also: https://github.com/olikraus/u8g2/issues/1472

closed 2021-05-09 by olikraus