olikraus/u8g2 #1409: U8x8 on Adafruit OLED

MIDIMinion, 2021-02-17:
Hi there,

Has this library been tested on Adafruit's 1.3" OLED display via I2C? I was using it for a project that had a SH1106 OLED, but I had to swap to a SSD1306 display and now I can't seem to get the display to work with this library. I've got the display working with the Adafruit 1306 example code, but I'd prefer to use U8x8 since my display needs are pretty basic.

Any insight would be much appreciated!
Thanks

olikraus, 2021-02-17:
I have not tested this specific Oled, but in general SSD1306 Oldest are supported.

MIDIMinion, 2021-02-17:
Am I correct in thinking that the following constructor should work with an Adafruit SSD1306 I2C display?

U8X8_SSD1306_128X64_NONAME_HW_I2C display8(19, 18 , U8X8_PIN_NONE);

I'm using a Teensy 4.0, so the clock is on pin 19 and the data is on pin 18. So far, I haven't been able to get this sketch to work with these displays. I've attached the whole sketch, incase it's not the constructor as I suspect.

Thanks again for your help!

[Teensy_4.0_SSD1306_tests.txt](https://github.com/olikraus/u8g2/files/5997192/Teensy_4.0_SSD1306_tests.txt)

MIDIMinion, 2021-02-18:
Hello again! a helpful internet stranger and I found a solution to this problem I was having. We discovered that the I2C address needed to be set manually to 0x3D and then bit shifted back one bit. The following line of code, run before `display8.begin();` got my Adafruit display up and running:

`display8.setI2CAddress(0x3D<<1);`

Thanks for your help, I hope this is helpful to this really excellent library, and I hope anyone who's stuck on the same problem can find this solution!

olikraus, 2021-02-18:
Great, nevertheless, I am confused by your constructor.
It should be instead a SW constructor:
U8X8_SSD1306_128X64_NONAME_SW_I2C display8(19, 18 , U8X8_PIN_NONE);
The HW constructor just requires the reset line (if any) as argument
U8X8_SSD1306_128X64_NONAME_SW_I2C display8(U8X8_PIN_NONE);

MIDIMinion, 2021-02-18:
I see, just tried it with SW and you're right it also works. Maybe because I'm using the default I2C pins on the Teensy, the HW constructor defaults to those pins?

olikraus, 2021-02-18:
Actually the HW I2C constructor ignores the data / clk pin on most systems. In fact the first argument is assumed to be the reset pin, which means, u8g2 will try to use 19 as reset output.
Better use 

U8X8_SSD1306_128X64_NONAME_HW_I2C display8(U8X8_PIN_NONE);

Instead for HW I2C

closed 2022-03-04 by olikraus