olikraus/u8g2 #1411: Problem using getBufferSize() and setBufferPtr()

urbantigerau, 2021-02-21:
Hi,
After a lot of Internet research on this I am still lost. This should "just work". 

I have  a project which uses 2x U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI OLED displays. There is no problem with the hardware, each display can be made to work both individually and together. I want different contents on each display and ideally separate buffers for each display. This has been raised before by other users and I thought it was solved after reading those posts. My plan was to  use the exiting allocated buffer, define an array as the other buffer and switch between them using the setBufferPtr() functionality. The processor is a 1284P using Mightycore. I have more than enough memory available for a duplicated buffer.

Here's my code:
`#include <Arduino.h>
//#include <U8x8lib.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

#define display_reset_2         31
#define display_reset_1         30
#define display_dc              29
#define display_select_2        4
#define display_select_1        28

uint8_t *buf_1, *buf_2;
unsigned int buf_size; 
uint8_t buffer_2[128];

// instantiate the two displays
U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI u8g2_1(U8G2_R0, /* cs=*/ display_select_1, /* dc=*/ display_dc, /* reset=*/ display_reset_1);      // display 1
U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI u8g2_2(U8G2_R0, /* cs=*/ display_select_2, /* dc=*/ display_dc, /* reset=*/ display_reset_2);      // display 2

void display_initialise()
{
   u8g2_1.begin(); u8g2_2.begin();                    // initialise displays
   u8g2_1.clearBuffer();                              // clear the common buffer
   u8g2_1.sendBuffer();  u8g2_2.sendBuffer();         // now send the data to clear the displays
   u8g2_1.setPowerSave(1); u8g2_2.setPowerSave(1);    // display off by default
}

void setup() 
{
  Serial.begin(115200);
  
  display_initialise();
  buf_size  = u8g2_1.getBufferSize();                                   // if uncommented this line fails compile: error: 'class U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI' has no member named 'getBufferSize'
  buf_size  = u8g2_1.getBufferTileWidth() << 3;                // this works as expected
  Serial.println(buf_size);
  
  buf_1 = u8g2_1.getBufferPtr();                                          // get the address of buffer 1
  buf_2 = &buffer_2[0];                                                   // get the address of buffer 2

  u8g2_1.setPowerSave(0); u8g2_2.setPowerSave(0);                         // display on
  // using buffer 1
  u8g2_1.setDrawColor(1);                                                 // draw a small box in the top left corner of display 1
  u8g2_1.drawBox(3,7,25,15);
  u8g2_1.sendBuffer();                                                    
  u8g2_2.sendBuffer();                                                    // draw the same small box in the top left corner of display 2
  
  // switch to buffer 2 - well I was planning to
  u8g2_2.setBufferPtr(buf_2);                                           // if uncommented this line fails compile: error: 'class U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI' has no member named 'setBufferPtr'
}

void loop() 
{
  // loop is void
}`

I can't understand why  buf_1 = u8g2_1.getBufferPtr();  works but 
buf_size  = u8g2_1.getBufferSize(); and u8g2_2.setBufferPtr(buf_2); both fail compile.

I can find both of these in the documentation, but can't find an example of their usage anywhere. Searching u8g2.setBufferPtr yields few hits, some in either Chinese or Japanese, and I am an English speaker. Google translate on these tends to confirm that I am correct in my assumption that both of these pieces of functionality should work.

I feel that the answer to this is glaringly obvious, and I just can't see it.  Any useful ideas welcome.
Regards,

olikraus, 2021-03-07:
U8g2 will always share the buffer it you allocate two displays with the same constructor.

.setBufferPtr should work in general, but you must apply memory area which includes at least 128x64/8 bytes.
At least it should compile.

urbantigerau, 2021-03-07:
I understand why and agree that buffer sharing by default is a reasonable idea. That is debated in other posts.

 I saw my problem with the buffer size when I developed a slower workaround for this issue.  My error here, apologies. I found that error after I posted.

Here's what I did:

`void save_display_1()
{
  int i;
     for ( i = 0; i < u8g2_display_buffer_size; i++)
     {
         display_buffer_1[i] = *(u8g2_display_buffer + i);
     }
}

void save_display_2()
{
  int i;
     for ( i = 0; i < u8g2_display_buffer_size; i++)
     {
         display_buffer_2[i] = *(u8g2_display_buffer + i);
     }
}

void restore_display_1()
{
  int i;
     for ( i = 0; i < u8g2_display_buffer_size; i++)
     {
         *(u8g2_display_buffer + i) =  display_buffer_1[i];
     }
}

void restore_display_2()
{
  int i;
     for ( i = 0; i < u8g2_display_buffer_size; i++)
     {
         *(u8g2_display_buffer + i)  =  display_buffer_2[i];
     }
}`

This works but is slower because instead of just setting the buffer pointer with .setBufferPtr I am copying the required contents from the holding buffers (being  arrays  display_buffer_1 and  display_buffer_2) to and from the main buffer.

I am mystified why the lines:
buf_size = u8g2_1.getBufferSize(); 
yields the compile error: 'class U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI' has no member named 'getBufferSize
and
u8g2_2.setBufferPtr(buf_2);
yields compiler error class U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI' has no member named 'setBufferPtr

My original plan was to manipulate the pointers so that I could avoid some of the array copying.  While I have achieved the effect I wanted with the work-around the array copying is an overhead and slows the overall display update time. Manipulating pointers would be roughly twice as fast (I think).

Here's another code fragment with the errors directly from the Arduino IDE (1.8.7) using only 1 display for simplicity.
`#include <Arduino.h>
#include <U8g2lib.h>

#define display_reset_1         30
#define display_dc              29
#define display_select_1        28

uint8_t *buf_1, *buf_2;
uint8_t display_buffer_1[1024];

U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI u8g2_1(U8G2_R0, /* cs=*/ display_select_1, /* dc=*/ display_dc, /* reset=*/ display_reset_1);      // display 1

void setup() 
{
u8g2_1.begin();
u8g2_1.clearBuffer();                                          // clear the common buffer
u8g2_1.sendBuffer();

buf_1 = u8g2_1.getBufferPtr(); // get the address of buffer 1 as assigned by U8g2
buf_2 = &display_buffer_1[0];   // get the address of the array assigned as the new buffer
// now some code to change contents of display_buffer_1 as required, the details of which don't matter here - not shown
// now change the pointer to display that buffer
u8g2_1.setBufferPtr(buf_2);
u8g2_1.sendBuffer();
}

void loop() 
{
  // no loop in this case:
}`

and here is the compiler output:
C:\Users\fredl\AppData\Local\Temp\arduino_modified_sketch_706272\sketch_mar07b.ino: In function 'void setup()':

sketch_mar07b:23:8: error: 'class U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI' has no member named 'setBufferPtr'

 u8g2_1.setBufferPtr(buf_2);

        ^

exit status 1
'class U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI' has no member named 'setBufferPtr'

I remain mystified why this won't compile. Any assistance on the source of this error would be much appreciated.

Regards,

olikraus, 2021-03-07:
You need to uncomment this definition in your local copy of u8g2:
https://github.com/olikraus/u8g2/blob/7fc361f3d2ab218289114e9a6631e40b1143e773/csrc/u8g2.h#L81-L87

urbantigerau, 2021-03-07:
Thanks for that, reading the documentation it sounds a little dangerous. Which why it is off by default I guess.  It seems a reasonable way to implement the functionality. However, I don't mind a bit of danger. An opportunity to learn. I have a number of projects which use U8g2 in the standard "static" manner, and it is clear I will have to remember to check the status of this definition, or perhaps change every one of them to use the dynamic allocation method. 

I will try this and see what happens.
Regards,

B-ohemian, 2021-08-14:
I have the same problem with you,and now I solve it with your method ,thanks

closed 2022-03-04 by olikraus