olikraus/u8g2 #1413: Using U8g2 with an already-established I2C Wire/Wire1 Instance?

dstulken, 2021-02-26:
Hello!   I'm attempting to use U8g2 to drive an SSD1309 OLED display, from the second hardware I2C bus on an ESP32, in the Arduino environment.  My test code works fine when using the SW I2C method, so I know the hardware and wiring are OK.  But the display remains blank when using "2ND_HW".
  
I'm aware of Issue #779, "pin remapping for second I2C interface", and that it was put on hold due to the large effort to implement.  

Rather than re-opening #779 (or reviving the modify-every-constructor solution), I had a thought that it might be possible for U8g2 to instead simply skip the I2C bus initialization entirely, and utilize an already existing (and configured) Wire / Wire1 instance.   My project (and I assume many others) already has a "Wire1.begin(dataPin, clockPin, clockRate)" call at the beginning of the setup function to initialize that bus, which other I2C devices use, so the "extra" U8g2 initialization calls to Wire1.begin() and Wire1.setClock() seemed unnecessary in this case.  

I had intended to propose this as an alternate solution to #779. A global #define could be added that could allow any I2C display (2ND_HW or otherwise) to skip the I2C init and re-use an established Wire instance, sidestepping the labor-intensive mass constructor updates entirely, for all displays at once.  But in my testing, with the separate bus init in my code, and a modified U8g2 library with the Wire1 I2C inits commented out, I still only see a blank display. 

I attempted to dig deeper, but for a project as complex as this, had a bit of trouble tracing how the library works and/or what might be the problem.  Do you see any reason why this approach wouldn't work?  

Thanks in advance.

dstulken, 2021-02-26:
Ok, I found the issue - it was the missing WIRE_INTERFACES_COUNT definition causing the whole code block to not be executed at all.  I posted some info on that at the end of Issue #961.   

With that out of the way, I went back and did a test with all of the Wire1.begin() and Wire1.setClock() calls commented out, and a separate Wire1.begin(dataPin, clockPin, clockRate) call in my setup() function, and it works just fine.  I then commented out that begin(...) function in my setup() function, and it fails as expected (blank OLED), so using an "external" I2C setup is working.

So with that being the case, what do you think of this concept (a flag to disable I2C bus initialization in U8g2) as a global solution to #779?  It could actually apply to any I2C bus (Wire or Wire1), and would actually scale nicely with no changes needed if any future device supports additional (3+) hardware I2C busses.

bodgit, 2021-03-05:
I would also like to have this functionality. I have an I2C bus that has numerous devices on it, of which an OLED device is just one of them.

Also maybe related, is it possible to override the I2C address for the device as well? Usually I2C devices have some address select pin(s) that can change the address to one of 2 or 4 choices but these SSD1306-based OLED devices that I have don't seem to have any, but there are cheap I2C address translation IC's that can be used so you could in theory have multiples of the same OLED device on the same bus.

olikraus, 2021-03-07:
@dstulken 
The
`#if defined(U8X8_HAVE_2ND_HW_I2C) || defined(ARDUINO_ARCH_ESP32)`
will only work if the Wire1 object is always present for the ESP32. Is this the case?
I mean, will there be any issues for users who do not use Wire1 object, but u8g2 will have code with Wire1 object?
Will blank Arduino sketch compile without error on ESP32 if the above statement is added?

@bodgit 
There is u8g2.setI2CAddress https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress

closed 2022-03-04 by olikraus