olikraus/u8g2 #1414: SH1122 I2C

spiderjuka, 2021-02-26:
Hello,

I need to buy some display 256x64 OLED which can work o I2C with your library.
Can you give me any link to such kind of display where I can buy it?
I already buy one on aliexpress but it doesn't work:
[https://www.aliexpress.com/item/12832-2-08-inch-display-SSD1307-PM-OLED-256x64-SH1122-7pin-4-Wire-SPI-3-Wire/32865527538.html?spm=a2g0s.9042311.0.0.51495c0fKd2CyB](url)
Maich choice was: No.10 Module 1 25664 White SH1122 7Pin SPI/IIC

Thank you very much for support.
Best Regards.
Jerzy Dybcak

olikraus, 2021-03-07:
The link product has a very misleading documentation, it talks about a ssd1307 128x32 and a sh1122 256x64 at the same time. Strange.
I have ordered this display: https://aliexpress.com/item/32847888835.html

theeprawn, 2021-04-05:
I obtained a SH1122 for Midas Displays - 2.08" 256x64pixel
https://www.midasdisplays.com/product-explorer/oleds/cog-graphic-oleds/mcot256064d
Farnell stock them as well (in UK atleast) but cost more
https://uk.farnell.com/search?st=mcot256064d

You will need to make own interface PCB, to solder hotbar to, needs 5 capacitors and some resistors depending what your connecting. Logic uses 3v to 5v and OLED display 12v (8v to 20v) uses a current sense resistor, the value is calculated from the display voltage you going to use,

If you need to refresh the display more than 2fps, suggest you use SPI interface.

closed 2022-03-04 by olikraus