olikraus/u8g2 #1419: how can i print chinese？

qpmc1s, 2021-03-04:
platform: C esp32 esp-idf gb2312

The screen becomes black when I use the code below。
And， The same phenomenon happens when I change to other fonts。
How can I print Chinese with u8g2？

`u8g2_ClearBuffer(&u8g2);

u8g2_SetFont(&u8g2, u8g2_font_wqy16_t_gb2312);
u8g2_DrawUTF8( &u8g2, 0, 0, "啊") ;
u8g2_SendBuffer(&u8g2);`

olikraus, 2021-03-07:
The reference point is the lower left edge, so try this instead:
```
u8g2_DrawUTF8( &u8g2, 0, 16, "啊") ;
```

qpmc1s, 2021-03-07:
Thanks!
I change to "u8g2_DrawUTF8( &u8g2, 0, 16, "啊") ;" or "u8g2_DrawStr( &u8g2, 0, 16, "啊") ;"

then, the character"。" on the screen instead of “啊”

my editor‘s encoding is "gb2312"

What might be wrong with me?

Thanks!Thanks!Thanks!

olikraus, 2021-03-07:
You need to use UTF-8 encoding in your editor. U8g2 expects all strings to be encoded in UTF-8 (that's why the function is called DrawUTF8)

qpmc1s, 2021-03-07:
Thank you. 
It works.

closed 2021-03-07 by qpmc1s