olikraus/u8g2 #1420: U8g2 and arduino-esp32 version 1.0.5 error

cbpdk, 2021-03-05:
Arduino ide 1.8.13
U8g2 version 2.28.10
arduino-esp32 version 1.0.5
TTGO T8 ESP32 V1.8
Cpu: 40 Mhz
Flash: 40 Mhz
4 Mb flash
Huge app
PSRAM disabled

I did some development using arduino-esp32 version 1.0.4 and U8g2 to display data on a ST7565, all working.
Having left it for some time, I updated the arduino-esp32 to version 1.0.5, recompiled and the esp32 got ino a reboot loop.
Commenting out all U8g2, it did not reboot, but executed.
Thinking the TTGO was damaged, I tried another esp32, same result.
Higher cpu frequency did not help.

Trying a simple Hello World sketch also failed!

Here is the Hello World sketch used:
`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

#include <time.h>
#include <sys/time.h>

U8G2_ST7565_JLX12864_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 18, /* data=*/ 23, /* cs=*/ 25, /* dc=*/ 33, /* reset=*/ 32);

#define uS_TO_MS_FACTOR 1000

void setup(void) {
  Serial.begin(115200);
  Serial.println(F("LOADING....."));
  Serial.flush();
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();         // clear the internal memory
  u8g2.setFont(u8g2_font_t0_18b_mf);
  u8g2.setContrast(172);
  u8g2.drawStr(0, 10, "Hello World!"); // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  esp_sleep_enable_timer_wakeup(1000 * uS_TO_MS_FACTOR); //x seconds
  esp_light_sleep_start();
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
}`

Compiling using arduino-esp32 version 1.0.4, it executes with no errors.

The result using arduino-esp32 version 1.0.5 causes the following with approx. 10 seconds interval (TTGO):
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6360
entry 0x400806b4
`

olikraus, 2021-03-07:
Hmm... not sure how I can support here.

cbpdk, 2021-03-07:
Thank you, I was not sure whether to raise the issue here or in the arduino-esp32 github.
As it appears to be a change in the arduino-esp32 package, I shall copy the issue to there.

cbpdk, 2021-03-13:
Resolved, error 40!
I had forgotten that I had added code to set up SPI1 for ESP32,  in short adding this to U8x8lib.cpp:
/* Declare a second SPI in case of ESP32 */
#if defined(ARDUINO_ARCH_ESP32)
#if SPI_INTERFACES_COUNT > 1
SPIClass SPI1 = SPIClass(HSPI);
#endif
#endif
Strange that it worked with the arduino-esp32  v1.0.4  when added to U8g2, but not with v1.0.5.

Adding
SPIClass SPI1 = SPIClass(HSPI); 
to the main sketch does not cause errors using v1.0.5.

closed 2022-03-04 by olikraus