olikraus/u8g2 #1425: SSD1322 256x64 display not working

elmahdi-lab, 2021-03-09:
Hi,

I received today an oled display (SSD1322 256x64) that I bought from [https://www.ebay.com/itm/301231209854](eBay). I'm using an Arduino Mega with the following wiring:
OLED    ->     MEGA
1 GND  ->     GND
2 VCC   ->     3.3V
4 DC     ->     Pin 9
7 SCL    ->     SCLK Pin 13
8 SDA   ->     MOSI Pin 11
15 CS   ->      Pin 10
16 RES ->      Pin 8

From what I can see from the PCB, R19 & R20 & & R21 & R22 are all not connected. The seller indicated in the product page that the interface is set to 8-bit 8080 parallel by default.

I've tried example sketchs with these constructors (one by one):
`U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	// Enable U8G2_16BIT in u8g2.h`

I enabled U8G2_16BIT, while I was looking around I read that the arduino can't use full buffer?! So in the constructor someone suggested to use U8G2_R3.

I've done all that but the display is not even turning on one pixel. I'm clueless to whats happening, either I got the wrong product or it's not working or I've made an error. But I can't see where?

I would really appreciate your help.

Thanks.

elmahdi-lab, 2021-03-09:
I asked the seller if he knew what was going on, he suggested I move the R19 to R18 in order to activate the 4W SPI interface, I did it, but still no joy!

olikraus, 2021-03-09:
Yes, R19 needs to be moved to R18 location

Did you use U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8) for testing? 

Did you follow the instructions from here?
https://www.buydisplay.com/arduino/Interfacing_ER-OLEDM028-1_to_Arduino.pdf
You need to tie several lines to ground!

elmahdi-lab, 2021-03-10:
Hi,
Thank you for your answer, I made an error in my last message the seller suggested I move the resistor from R18 to R19, because before it was in R18, according to him the display was in 8080 parallel mode, and making the resistor move to R19 would put the display 4W SPI mode.

In the PDF you provided, they say for the 4W SPI R19 and R23 should be connected while R18 and R20 not connected. Why would I move back R19 to R18?

[image]

olikraus, 2021-03-10:
You could also check this document: https://www.buydisplay.com/download/interfacing/ER-OLEDM028-1_Interfacing.pdf
According to your image, your current setup is displayed in the upper left (R23 and R18), which is 8080 parallel.
But you need 4 wire SPI, which is shown in the lower left (R23 and R19). So you need to move the Resistor from R19 to R18. In other words, there must be a 0 Ohm resistor at R23 and R18 position, while R20 and R19 must be left empty.

closed 2021-05-09 by olikraus