olikraus/u8g2 #1426: STM32F7 and ST7528 LCD

s3chand9, 2021-03-12:
**Hey There,

I'm new to u8g2 library and trying to bring up 128x128 LCD(G128ALGFGSW6WTC3XAM) using i2c. I'm seeing data is being sent from i2c but nothing comes on LCD display. Can you please tell me correct way to implement u8g2 lib using i2c. I'm struggling to send data on LCD. Any help would be greatly appreciated.** 

/* USER CODE BEGIN Header */
/* Includes ------------------------------------------------------------------*/
#include "main.h"
#include "stm32f7xx_hal.h"

/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
#include "u8g2.h"
#include "u8x8.h"
/* USER CODE END Includes */

/* Private variables ---------------------------------------------------------*/

ETH_HandleTypeDef heth;

I2C_HandleTypeDef hi2c1;

UART_HandleTypeDef huart3;

PCD_HandleTypeDef hpcd_USB_OTG_FS;

/* USER CODE BEGIN PV */

u8x8_t u8x8;                    // u8x8 object
u8g2_t u8g2;
/* Private variables ---------------------------------------------------------*/
#define I2C_ADDRESS 0x3D
//#define I2C_ADDRESS 0x7A
/* USER CODE END PV */

/* Private function prototypes -----------------------------------------------*/
void SystemClock_Config(void);
static void MX_GPIO_Init(void);
static void MX_I2C1_Init(void);
void u8g2_Setup_ssd1306_i2c_128x64_noname_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
//void u8g2_Setup_st7528_erc16064_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb); //u8g2_Setup_st7528_i2c_nhd_c160100_f
//void u8g2_Setup_st7571_128x128_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb); u8g2_Setup_st7528_erc16064_f
/* USER CODE BEGIN PFP */

/* USER CODE END PFP */

/* Private user code ---------------------------------------------------------*/
/* USER CODE BEGIN 0 */

/*need to update from below*/
uint8_t u8x8_byte_stm32hal_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	static uint8_t buffer[32];		/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
	static uint8_t buf_idx;
	uint8_t *data;

  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
    {
        data = (uint8_t *)arg_ptr;
        while( arg_int > 0 )
        {
			buffer[buf_idx++] = *data;
			data++;
			arg_int--;
        }
    }
    	break;
    case U8X8_MSG_BYTE_INIT:
    	break;
    case U8X8_MSG_BYTE_SET_DC:
    	break;
    case U8X8_MSG_BYTE_START_TRANSFER:
		{
			buf_idx = 0;
		}
		break;
    case U8X8_MSG_BYTE_END_TRANSFER:
	{
		uint8_t iaddress = I2C_ADDRESS;
		//HAL_I2C_Master_Transmit_DMA(&hi2c1, (uint16_t)iaddress<<1, &buffer[0], buf_idx);
		HAL_I2C_Master_Transmit(&hi2c1, (uint16_t)iaddress<<1, &buffer[0], buf_idx, 20u);
		//TODO Investigate why delay is needed here.
		//Seems like DMA feeding bytes too fast.
		volatile uint32_t i;
		for (i = 1; i <= 500; i++);
	}
    	break;
    default:
      return 0;
  }
  return 1;
}

uint8_t psoc_gpio_and_delay_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  switch(msg)
  {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:
      /* only support for software I2C*/

      break;
    case U8X8_MSG_DELAY_NANO:
      /* not required for SW I2C */
    {
		volatile uint32_t i;
		for (i = 1; i <= arg_int*10; i++);
    }
      break;

    case U8X8_MSG_DELAY_10MICRO:
      /* not used at the moment */
      break;

    case U8X8_MSG_DELAY_100NANO:
      /* not used at the moment */
      break;

    case U8X8_MSG_DELAY_MILLI:
      HAL_Delay(arg_int);
      break;
    case U8X8_MSG_DELAY_I2C:
      /* arg_int is 1 or 4: 100KHz (5us) or 400KHz (1.25us) */
      //delay_micro_seconds(arg_int<=2?5:1);
      break;

    case U8X8_MSG_GPIO_I2C_CLOCK:

      break;
/*
    case U8X8_MSG_GPIO_MENU_SELECT:
      u8x8_SetGPIOResult(u8x8, Chip_GPIO_GetPinState(LPC_GPIO, KEY_SELECT_PORT, KEY_SELECT_PIN));
      break;
    case U8X8_MSG_GPIO_MENU_NEXT:
      u8x8_SetGPIOResult(u8x8, Chip_GPIO_GetPinState(LPC_GPIO, KEY_NEXT_PORT, KEY_NEXT_PIN));
      break;
    case U8X8_MSG_GPIO_MENU_PREV:
      u8x8_SetGPIOResult(u8x8, Chip_GPIO_GetPinState(LPC_GPIO, KEY_PREV_PORT, KEY_PREV_PIN));
      break;

    case U8X8_MSG_GPIO_MENU_HOME:
      u8x8_SetGPIOResult(u8x8, Chip_GPIO_GetPinState(LPC_GPIO, KEY_HOME_PORT, KEY_HOME_PIN));
      break;
*/
    default:
      u8x8_SetGPIOResult(u8x8, 1);
      break;
  }
  return 1;
}/*updated code ends here*/
/* USER CODE END 0 */

/**
  * @brief  The application entry point.
  * @retval int
  */
int main(void)
{
  /* USER CODE BEGIN 1 */
  static uint8_t testing = 0U;
  static uint8_t i = 0U;
  /* USER CODE END 1 */

  /* MCU Configuration--------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* USER CODE BEGIN Init */

  /* USER CODE END Init */

  /* Configure the system clock */
  SystemClock_Config();

  /* USER CODE BEGIN SysInit */

  /* USER CODE END SysInit */

  /* Initialize all configured peripherals */
  MX_GPIO_Init();
  //MX_ETH_Init();
  //MX_USART3_UART_Init();
  //MX_USB_OTG_FS_PCD_Init();
  MX_I2C1_Init();
/*   for(i=0; i<128; i++) {
	  uint8_t pData[] = "hello";
	  //HAL_StatusTypeDef iicstatus;
	  testing = HAL_I2C_Master_Transmit(&hi2c1, i, pData, sizeof(pData), 10);
	  if (testing == HAL_OK)
		  {
			  test = 1U;
		  }
	  else
	  {
		  test = 0U;
	  }
  } */

  /* USER CODE BEGIN 2 */
#ifdef USE_U8X8
  u8x8_Setup(&u8x8,
		  u8x8_d_ssd1306_128x64_noname,
		  u8x8_cad_ssd13xx_i2c,
		  u8x8_byte_stm32hal_hw_i2c,
		  psoc_gpio_and_delay_cb);
  
    u8x8_Setup(&u8x8,
		  u8x8_d_ssd1306_128x64_noname,
		  u8x8_cad_ssd13xx_i2c,
		  u8x8_byte_stm32hal_hw_i2c,
		  psoc_gpio_and_delay_cb);

  u8x8_InitDisplay(&u8x8);
  u8x8_SetPowerSave(&u8x8, 0);
  u8x8_ClearDisplay(&u8x8);
  u8x8_SetFont(&u8x8, u8x8_font_amstrad_cpc_extended_r);
#else
 u8g2_Setup_ssd1306_i2c_128x64_noname_f(&u8g2,
		  U8G2_R0,
		  u8x8_byte_stm32hal_hw_i2c,
		  psoc_gpio_and_delay_cb);
//   u8g2_Setup_st7528_erc16064_f(&u8g2,
//		   U8G2_R0,
//		   u8x8_byte_stm32hal_hw_i2c,
//		   psoc_gpio_and_delay_cb);
//    u8g2_Setup_st7528_i2c_erc16064_1(u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
  u8g2_SetI2CAddress(&u8g2, I2C_ADDRESS*2);
  //u8g2_SetI2CAddress(&u8g2, I2C_ADDRESS);
  
  u8g2_InitDisplay(&u8g2);
  u8g2_SetPowerSave(&u8g2, 0);
  u8g2_ClearDisplay(&u8g2);

  u8g2_SetDrawColor(&u8g2, 1);
  u8g2_SetFontDirection(&u8g2, 0);
#endif
  /* USER CODE END 2 */

  /* Infinite loop */
  /* USER CODE BEGIN WHILE */
  while (1)
  {
    /* USER CODE END WHILE */

  /* USER CODE BEGIN 3 */
#ifdef USE_U8X8
	  u8x8_ClearDisplay(&u8x8);
      u8x8_DrawString(&u8x8, 0, 0, "Hello World!");
      HAL_Delay(500u);
#else
		u8g2_ClearBuffer(&u8g2);

		u8g2_SetFontMode(&u8g2, 0);	// Transparent
		u8g2_SetFontDirection(&u8g2, 0);
		u8g2_SetFont(&u8g2, u8g2_font_inb24_mf);
		u8g2_DrawStr(&u8g2, 0, 30, "U");

 		u8g2_SetFontDirection(&u8g2, 1);
		u8g2_SetFont(&u8g2, u8g2_font_inb30_mn);
		u8g2_DrawStr(&u8g2, 21,8,"8");

		u8g2_SetFontDirection(&u8g2, 0);
		u8g2_SetFont(&u8g2, u8g2_font_inb24_mf);
		u8g2_DrawStr(&u8g2, 51,30,"g");
		u8g2_DrawStr(&u8g2, 67,30,"\xb2");

 		u8g2_DrawHLine(&u8g2, 2, 35, 47);
		u8g2_DrawHLine(&u8g2, 3, 36, 47);
		u8g2_DrawVLine(&u8g2, 45, 32, 12);
		u8g2_DrawVLine(&u8g2, 46, 33, 12);
		
		//u8g2_DrawHLine(&u8g2, 100,100,125);
		//u8g2_DrawHLine(&u8g2, 60, 57, 17);
		//u8g2_DrawVLine(&u8g2, 65, 37, 12);
		//u8g2_DrawVLine(&u8g2, 56, 30, 12); 

		u8g2_SetFont(&u8g2, u8g2_font_4x6_tr);

		//u8g2_DrawStr(&u8g2, 1,54,"github.com/olikraus/u8g2");

		u8g2_SendBuffer(&u8g2);
#endif
      HAL_Delay(2000);
  }
  /* USER CODE END 3 */
}

/**
  * @brief I2C1 Initialization Function
  * @param None
  * @retval None
  */
static void MX_I2C1_Init(void)
{

  /* USER CODE BEGIN I2C1_Init 0 */

  /* USER CODE END I2C1_Init 0 */

  /* USER CODE BEGIN I2C1_Init 1 */

  /* USER CODE END I2C1_Init 1 */
  hi2c1.Instance = I2C1;
  hi2c1.Init.Timing = 0x20303E5D;
  hi2c1.Init.OwnAddress1 = 0;
  hi2c1.Init.AddressingMode = I2C_ADDRESSINGMODE_7BIT;
  hi2c1.Init.DualAddressMode = I2C_DUALADDRESS_DISABLE;
  hi2c1.Init.OwnAddress2 = 0;
  hi2c1.Init.OwnAddress2Masks = I2C_OA2_NOMASK;
  hi2c1.Init.GeneralCallMode = I2C_GENERALCALL_DISABLE;
  hi2c1.Init.NoStretchMode = I2C_NOSTRETCH_DISABLE;
  if (HAL_I2C_Init(&hi2c1) != HAL_OK)
  {
    Error_Handler();
  }
  /** Configure Analogue filter
  */
  if (HAL_I2CEx_ConfigAnalogFilter(&hi2c1, I2C_ANALOGFILTER_ENABLE) != HAL_OK)
  {
    Error_Handler();
  }
  /** Configure Digital filter
  */
  if (HAL_I2CEx_ConfigDigitalFilter(&hi2c1, 0) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN I2C1_Init 2 */

  /* USER CODE END I2C1_Init 2 */

}

/************************ (C) COPYRIGHT STMicroelectronics *****END OF FILE****/

olikraus, 2021-03-14:
I do not know our HAL, but from u8g2 perspective it looks ok.
Of course the SSD1306 setup will not work with the ST7528 setup (which seems to be commented)

s3chand9, 2021-03-14:
Hey,

Thank you for the response. I tried using ST7528 setup.. its transmitting data over i2c but I'm only getting Negative acknowledgment. I tried using different slave address and setup but no change in acknowledgement. All the time its throwing HAL_Error return. Do you know up to your best of your knowledge what could be the reason behind this or if I'm doing something wrong here? Any help would be great.. Thank you.

s3chand9, 2021-03-14:
> I do not know our HAL, but from u8g2 perspective it looks ok.
> Of course the SSD1306 setup will not work with the ST7528 setup (which seems to be commented)

Hey,

Thank you for the response. I tried using ST7528 setup.. its transmitting data over i2c but I'm only getting Negative acknowledgment. I tried using different slave address and setup but no change in acknowledgement. All the time its throwing HAL_Error return. Do you know up to your best of your knowledge what could be the reason behind this or if I'm doing something wrong here? Any help would be great.. Thank you.

olikraus, 2021-03-14:
Maybe the iwc pull-up resistors are missing.

s3chand9, 2021-03-15:
> Maybe the i2c pull-up resistors are missing.

Board does have proper pull up resistors connected. But still its not getting acknowledgment from slave device.

olikraus, 2021-05-09:
Not sure how to continue. Can we close this issue?

closed 2022-03-04 by olikraus