olikraus/u8g2 #1428: Can't drive OLED when using SD1S306_128X64_NONAME_SW_I2C

mogerr, 2021-03-13:
Can't drive SD1306 OLED when using routines likes "Hello World"

`U8X8_SD1S306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ 9, /* data=*/ 10, /* reset=*/ U8X8_PIN_NONE); 
	      // Digispark ATTiny85
`

The same code can be driven when **esp8266-12E** development board with **cp2102** is used, but cannot be driven when **esp8266-12F** development board with **cp2102** or **CH340** is used. It's strange and maddening.

`scl=9, sda=10` comes as a software driver pin for OLED.

Has anyone encountered the same problem?

olikraus, 2021-05-09:
Not sure how to continue?
Can I close this issue?

closed 2022-03-04 by olikraus

yukiloh, 2023-09-11:
Yes, I have encountered the same problem.
When using ESP8266 and CH340, I encountered a problem of USB disconnection. After the program is uploaded, the device disconnects from the computer. And it happens periodically, constantly losing connection and reconnecting.
I'm sure this is a software issue rather than a hardware problem because I uploaded other test codes without experiencing any disconnection.

olikraus, 2023-09-13:
Still I don't know how to support here. 
The only idea which I have, could be to look into the watchdog timer of the ESP OS Maybe turn off the watchdog completly (google comes up with this suggestion: `ESP.wdtDisable();`)