olikraus/u8g2 #1433: New HX1230 display, do not work set_Contrast.

Davimoyes, 2021-03-16:
Hello everyone!
I'm using new HX1230 display, bought actually, 
https://es.aliexpress.com/item/4001352938776.html?spm=a2g0s.9042311.0.0.274263c0PzD9IN

They work, and start perfectly and they display "Hello World" but you do not set_Contrast neither ArduinoLowPower.h library do not works to wake up the display.
I've tried changing everything and it doesn't work.9

I have tried with another display bought last year and YES, it works. 
Please, do you know where can the fault be, fault Hardware 
or comands library changed?.

Thanks you very much for your best support.

This is the code:

#include <Arduino.h>
#include <U8g2lib.h>
#include <ArduinoLowPower.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif

U8G2_HX1230_96X68_F_3W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* reset=*/9 );

int pin = 23;
unsigned long tempo_sleep = 0;
int period = 10000;

void setup(void) {
  
  pinMode(pin,INPUT_PULLUP);
  u8g2.begin();
  LowPower.attachInterruptWakeup(pin, pinInterrupt, CHANGE); 
}

void pinInterrupt(){
 u8g2.begin();
 u8g2.setPowerSave(0);
 u8g2.setContrast(125); 
}

void loop(void) {
  
  u8g2.clearBuffer();					
  u8g2.setFont(u8g2_font_ncenB08_tr);	
  u8g2.drawStr(0,30,"Hello World!");	
  u8g2.sendBuffer();					
  delay(1000);
  
  if(millis() - tempo_sleep > period){ 
  tempo_sleep = millis();
  u8g2.setPowerSave(1);
  LowPower.sleep();        
  }
  
}

spase74, 2021-05-07:
From my experience, some of Nokia1202/HX1230 displays simply don't support contrast settings. Tried to communicate to them on low level without success.

olikraus, 2021-05-09:
So I assume, this issue can be closed, right?

spase74, 2021-05-09:
Yes, there's no way to make bias adj commands that are unsupported in some particular items to work.

closed 2021-05-09 by olikraus