olikraus/u8g2 #1448: u8g2 & SparkFun Artemis

adamgarbo, 2021-03-28:
Hello,

I am attempting to troubleshoot an issue with an [SSD1306 OLED](https://www.sparkfun.com/products/17153) and the [SparkFun Artemis](https://www.sparkfun.com/products/16401) processor. From searching through the repository, it doesn't look like there have previously been any Artemis-related issues.

The problem I'm facing is that the OLED will simply either turn off after a short period of time (30 s to 3 min), or produce random pixel noise and stop responding. 

I am communicating with the display over I2C and using the following constructor:
`U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);`

Data is printed to the display every second with a simple function:

```
void printImu() 
{
u8g2.clearBuffer();
u8g2.setCursor(0, 10);
u8g2.print(roll);
u8g2.setCursor(0, 20);
u8g2.print(pitch);
u8g2.setCursor(0, 30);
u8g2.print(yaw);
u8g2.sendBuffer();
}
```

I'm curious if anyone has experienced any similar issues in the past, or may have an idea of what could be happening.

The OLED's part number is [QG-2832TLBFG02](https://datasheet.lcsc.com/szlcsc/1809291515_Shenzhen-Allvision-Tech-QG-2832TLBFG02_C91220.pdf) made by Shenzhen Allvision Tech if that's useful information.

Cheers,
Adam

olikraus, 2021-05-09:
Not sure how I can help here. Hope you don't mind if I close this issue once...

adamgarbo, 2021-05-09:
Hi @olikraus,

Happy to close this issue as unresolved. I was hoping that someone in the community may have experienced a similar issue, but the SparkFun Artemis doesn't seem to like to play with OLEDs or RGB LEDs (e.g. FastLED).

Thanks for a great library!

Cheers,
Adam

closed 2021-05-09 by adamgarbo