olikraus/u8g2 #1450: SSD10306 I2C ESP-IDF Crashes

m0thm, 2021-03-29:
Using esp32 esp-idf on DOIT DevKitC

SCL_GPIO = 22
SDA_GPIO = 21
MB_OLED_ADDRESS = 0x78

> ESP_LOGI(TAG, "SETUP: I2C: MB: SSD1306");
                                u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT; // Structure for u8g2 on ESP32
                                    u8g2_esp32_hal.sda = SDA_GPIO;                        // Set SDA Pin
                                    u8g2_esp32_hal.scl = SCL_GPIO;                        // Set SCL Pin
                                    u8g2_esp32_hal_init(u8g2_esp32_hal);                  // Init u8g2 for ESP32
                                u8g2_t u8g2;                                              // A Structure which will contain all the data for one display
                                u8g2_Setup_ssd1306_128x64_noname_f(
                                   &u8g2,
                                   U8G2_R0,
                                   u8g2_esp32_i2c_byte_cb,
                                   u8g2_esp32_gpio_and_delay_cb);                         // Initlize u8g2
                                u8x8_SetI2CAddress(&u8g2.u8x8, MB_OLED_ADDRESS);
                                u8g2_InitDisplay(&u8g2);

As soon as  u8g2_InitDisplay(&u8g2); is triggered I get an error

> E (399) i2c: i2c_param_config(644): i2c clock choice is invalid, please check flag and frequency
E (409) err: esp_err_t = 258
assertion "0 && "i2c_param_config(I2C_MASTER_NUM, &conf)"" failed: file "../main/u8g2_esp32_hal.c", line 129, function: u8g2_esp32_i2c_byte_cb

The uController then reboots and basically goes nuts. The only way i get to see anything is to kill the monitor so i can scroll back. I have poored over the web and go nowhere. OLED and uController work fine using the Arduino IDE & Library so the Hardware is good to go. The code compiles and everything resolves so my CMakeLists.txt etc seem to be good. I am sure I am missing something obvious.

olikraus, 2021-05-09:
Maybe there is a watchdog issue.

cbaurtx, 2021-07-13:
It has been a while. Have you been able to fix the issue?
I have the same problem, but found that it works when setting the LOG level to debug

ingamg, 2021-08-11:
Hello, add add the next instuction
			memset(&conf, 0, sizeof(i2c_config_t)); 
after
                        i2c_config_t conf;
and solve de problem.

Best Regards.
Alejandro

christhomas, 2024-08-10:
@ingamg but can you explain why this fixed the problem for you?