olikraus/u8g2 #1453: Driver for uc1628

witc, 2021-03-31:
Dear Olikraus,

I would like to use LCD with driver uc1628 (160x160). Unfortunatelly I have only [this datasheet ](https://github.com/olikraus/u8g2/files/3928233/UC1628c_A1.0.pdf) 
Now I use LCD with uc1611 and it works very well. Probably the simpliest way is to modifiy uc1611 to uc1628?!. 
Just trying to find out what needs to be modified. 
I will be appriciate for any help.

Thank you very much. Jan.

witc, 2021-03-31:
@olikraus  I think the problem could be in DC logic level. From those which are listed in u8x8_cad.c I think the u8x8_cad_011 variant is the closest. But as you can see [in datasheet](https://github.com/olikraus/u8g2/files/3928233/UC1628c_A1.0.pdf) page 13, the WriteDataByte is double-byte command.
I come from cg160160_f  (uc1611) and there WriteDataByte is only one-byte.
I am not sure how to implemet the  WriteDataByte as double-byte command. 
If my thought are OK ?!?

Thank you very much. Jan.

witc, 2021-04-01:
@olikraus I edited uc1611 (cg160160_f) to uc1628 (160x160). Now I can see the text in LCD but:
1) The text is shifted to side (X pos)
2) When I repeatedly draw to LCD the same text, the text is every writing shifted in Y pos about 20 px.

What I edited towards to uc1611:
**1) Init sequence**
- not sure if CAA (30 and 31) Set Partial Display Start should set, if yes which value?
- not sure how to Set Scroll Line (I have 0)

**2) u8x8_display_info_t**
in bracket my values
 - which I am not sure: tile_width(20), tile_height(20), default_x_offset(160), flipmode_x_offset(0)

**3) added  doubled commad**
Because the uc1628 has doubled Byte write, I added  _u8x8_cad_SendCmd(u8x8, 0x001 );_to _u8x8_d_uc1611_common_ before drawing tile

I will be appreciate any help.

Best regards, Jan.

witc, 2021-04-06:
No it seems it works fine: [LCD]([url=https://postimg.cc/MXfrFs7G][img]https://i.postimg.cc/HW6DjRjb/IMG-20210406-125019.jpg[/img][/url])

But I had to set `/* default_x_offset = */ 48, in u8x8_uc1611_cg160160_display_info`  struct by experimental method.

I think there is probably some another mistake and I covered it by this offset. 
Any idea?

Best regards, Jan.

olikraus, 2021-04-18:
looks good. The offset should be fine.

olikraus, 2021-05-09:
Are you able to attach the modified code here? I could then try to add this to u8g2. 
Thanks a lot.

witc, 2021-05-10:
Hi,

The code I modified works, but is not completely clean. Will you clean it
yourself? If not, I can try it if I will have a time.
Thanks. Jan.

olikraus, 2021-05-10:
I can try to do the cleanup.

olikraus, 2021-09-02:
Not sure how to continue here, moving this to 2.30

olikraus, 2022-01-07:
Any feedback/progress?

witc, 2022-01-08:
Hi, I had some personal problems. Now i think I could be ready for finish
it during a weeks.

olikraus, 2022-01-08:
No issues, take your time

closed 2022-03-04 by olikraus

ullahsh, 2022-11-16:
Hi olikraus,
I am developing parallel port 8080 driver for UC1628 lcd, do you have any basic driver for parallel port to up the lcd.
wanted to know the state of WR0,WR1,RST and RS pin of LCD to write data into it.
Any help would be appreciated.
Regards,
Shafi

olikraus, 2022-11-16:
Continued here: #2011