olikraus/u8g2 #1459: 2nd hardware i2c with SSD1306 and a Teensy 4.0 (Arduino IDE)

Maxfojtik, 2021-04-09:
I am trying to run two 128x32 SSD1306 displays. When I was working with just one I had no problems. But then I switched "U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C" to "U8G2_SSD1306_128X32_UNIVISION_F_2ND_HW_I2C" and it does not write anything to the display. The Scanner example does find both of the displays at the correct addresses, so it is not wiring. I tried to look around the lib myself but sadly I could not completely understand it. It is very organized all things considered.

olikraus, 2021-04-18:
The second HW I2C is not supported on all devices.

Maxfojtik, 2021-04-23:
If anyone else is having this problem you can fix it by going into U8x8lib.cpp, Control-Fing "Wire1" go into the function around line 1350 and removing any line that has a # in it within that method. It seems like the library does not realize the teensy's actually have more than 1 Wire object but you can force it.

closed 2021-04-23 by Maxfojtik

spl01tprivate, 2023-07-18:
Thanks, working fine on ESP32-S3-Wroom-1 as well!