olikraus/u8g2 #1460: warning: deleting object of polymorphic class type

blazoncek, 2021-04-11:
Hi,
I'm trying to provide dynamic (runtime) selection (and change) of u8x8 display driver.
While creating new instance poses no problems, I do get compiler warning when deleting them.
The issue seem to be that the base class does not provide a virtual destructor.

Example:
if (type==SSD1306) u8x8 = (void *) new U8X8_SSD1306_128X32_UNIVISION_HW_I2C(U8X8_PIN_NONE, sclPin, sdaPin);
...
if (type==SSD1306) delete (static_cast<U8X8_SSD1306_128X32_UNIVISION_HW_I2C*>(u8x8));

I did not dig into u8x8 code. Does this pose any problems? Stackoverflow posts warn that this kind of warnings can be related to memory leaks since derived class' destructor is not being called.
Cheers.

olikraus, 2021-04-18:
> The issue seem to be that the base class does not provide a virtual destructor.

Might be possible, but on the other hand, u8g2 does not allocate any memory, so i do not know what todo in the destructor.
All in all, U8g2 does not allocate memory, so there can not be any memory leaks.

blazoncek, 2021-04-18:
Oh!
That is nice to hear. It is just a warning to ignore then. ;)
What if you add a virtual keyword anyway?

olikraus, 2021-05-09:
Hmmm.. u8g2 and u8x8 are both derived from the Arduino Print class, which also does not have a virtual destructor (see here https://github.com/arduino/Arduino/pull/4466). Will it help if I add a virtual destructor to u8g2 / u8x8? My C++ knowledge is not good enough to rate this, but it feels old.

blazoncek, 2021-05-10:
Thank you for taking care.
Looks like this is an old issue which Arduino authors neglected to address due to 656 bytes of extra memory used.

nomis, 2021-05-10:
As long as you do not create a situation where the object is considered to be a `Print` when it is deleted then the correct destructors will still be called.

blazoncek, 2021-05-10:
> As long as you do not create a situation

OOP assumes no (or just minor) prior knowledge of base classes internals to derive your own. So this statement contradicts OOP principles. ;)

blazoncek, 2021-05-16:
@olikraus I have added virtual destructor to U8X8 class and it removed the compiler warning message without ill effects.
I have not checked the increase of code size, though.
I did look into Print class and it does not require any special handling on instance end-of-life, so both options are ok AFAIK.
Since there is no memory allocation going on during instance creation I think it is safe (albeit annoying) to leave the code as-is.
[image]

closed 2021-05-16 by blazoncek