olikraus/u8g2 #1469: Pixel scaling

AntoRoni, 2021-04-18:
I am making a code that draws pixels on the ST7920 display, and I was wondering if there is any way to scale these pixels, for example, the following code prints a small "H" on the screen, is there any way to make it bigger?

```
void loop()
{
  u8g2.firstPage();
  do 
  {
    u8g2.setDrawColor(2);
    u8g2.drawPixel(1,0);
    u8g2.drawPixel(1,1);
    u8g2.drawPixel(1,2);
    u8g2.drawPixel(2,1);
    u8g2.drawPixel(3,0);
    u8g2.drawPixel(3,1);
    u8g2.drawPixel(3,2);
  } while( u8g2.nextPage());  
  while(1);
}
```

olikraus, 2021-04-19:
You could use drawBox instead

AntoRoni, 2021-04-19:
Thanks I already solved the problem using drawLines, with a scale algorithm

closed 2021-05-09 by olikraus