olikraus/u8g2 #1473: EPSON S1D15721

loschwa, 2021-04-22:
I am trying to use the EPSON driver chip S1D15721 connected to a 240x64 LCD. As this chip is very similar to the EPSON S1D15E06, I was using the S1D15E06_160100_4W_SW_SPI constructor. The datasheet for the S1D15721 can be found here: http://datasheet.datasheetarchive.com/originals/library/Datasheets-ISS16/DSAIH000309343.pdf.

I am using the Arduino IDE with the ESP32 MCU and the "Hello World"-example but there is nothing diplayed although a simple test program using SW SPI with the initialisation:

    _lcd_cmd2(LCD_CMD_BUILTIN_OSC_FREQ,0x00);
    _lcd_cmd1(LCD_CMD_BUILTIN_OSC|1);
    _lcd_cmd2(LCD_CMD_LC_DRIVE_VOLTAGE,0x07);
    _lcd_cmd2(LCD_CMD_EL_VOLUME,0x10);
    _lcd_cmd2(LCD_CMD_PWR_CONTROL,0b11111);
    _lcd_cmd3(LCD_CMD_DUTY_SET,0x11,0x00);

    _lcd_cmd1(LCD_CMD_DISPLAY_REVERSE|0);
    _lcd_cmd1(LCD_CMD_DISPLAY_ALL_LIGHT|0);
    _lcd_cmd2(LCD_CMD_DISPLAY_MODE,0b11);
    _lcd_cmd2(LCD_CMD_GRAY_SCALE_PATTERN,0x36);
    _lcd_cmd1(LCD_CMD_COM_OUT_STATUS|1);
    _lcd_cmd2(LCD_CMD_DISPLAY_START_LINE,8);
    _lcd_cmd1(LCD_CMD_DISPLAY_OFFON|1);

is working fine.

It appears to me that following the initialisation sequence triggered by u8g2.begin(); the CS is kept HIGH although it should be pulled LOW for every data transfer.

Many thanks for your help.

loschwa, 2021-04-24:
I now have modified "u8x8_d_s1d15e06.c" to match the command numbers and required settings of the S1D15721 datasheet. I also modified the display info to match the physical properties of the display (i.e. 240x64). The display now works, however it seems that the columns 0-7 are hidden and the columns 233-239 are not accessible.

  u8g2.clearBuffer();                 // clear the internal memory
  u8g2.clearDisplay();
  
  u8g2.setFont(u8g2_font_t0_12_tf); // choose a suitable font
  u8g2.drawStr(8,9,"Hello World! x=8, y=9");  // write something to the internal memory
  u8g2.setFont(u8g2_font_fub20_tf); // choose a suitable font
  u8g2.drawStr(20,40,"Hello World!");  // write something to the internal memory
  u8g2.drawCircle(120, 32, 30, U8G2_DRAW_ALL);
  u8g2.drawFrame(8,0,232,64);
  u8g2.sendBuffer();                  // transfer internal memory to the display*/

results in this display output:

[image]

Did I miss to modify any parameter?

loschwa, 2021-04-24:
Here the relevant parts of "u8x8_d_s1d15e06.c":

/*================================================*/
/* s1d15721 240x64 */

static const u8x8_display_info_t u8x8_s1d15e06_160100_display_info =
{
  /* chip_enable_level = */ 0,	/* s1d15e06 has low active CS */
  /* chip_disable_level = */ 1,

  /* post_chip_enable_wait_ns = */ 10,	/* uc1608 datasheet, page 39, actually 0 */
  /* pre_chip_disable_wait_ns = */ 20,	/* uc1608 datasheet, page 39 */
  /* reset_pulse_width_ms = */ 1, 	/* uc1608 datasheet, page 42 */
  /* post_reset_wait_ms = */ 10, 	
  /* sda_setup_time_ns = */ 30,		/* uc1608 datasheet, page 41 */
  /* sck_pulse_width_ns = */ 65,	/* half of cycle time  */
  /* sck_clock_hz = */ 8000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 30,	/* uc1608 datasheet, page 39 */
  /* write_pulse_width_ns = */ 65,	/* uc1608 datasheet, page 39 */
  /* tile_width = */ 30,		/* width of 20*8=160 pixel (30*8 = 240) */
  /* tile_hight = */ 8,        /* height 8*8 = 64*/
  /* default_x_offset = */ 0,	/* reused as y page offset */
  /* flipmode_x_offset = */ 0,	/* reused as y page offset */
  /* pixel_width = */ 240,
  /* pixel_height = */ 64
};

static const uint8_t u8x8_d_s1d15e06_160100_init_seq[] = {

  U8X8_START_TRANSFER(),           	/* enable chip, delay is part of the transfer start */

									/* (Command no in datasheet) Description */

  U8X8_C(0xC4|1),       	    	/* (5) Common Output Status (Reverse) */
  U8X8_CA(0xA6, 0x01),            	/* (3) Display Normal Reverse (Normal) */
  U8X8_CA(0xA4, 0x00), 	           	/* (4) Display All Light (Normal) */
  U8X8_CAA(0x6D,0x10,0x02), 		/* (18) Duty Set Command, 
  U8X8_CA(0x66, 0x01),           	/* (15) Display Mode, Parameter 0 (0 = Gray Scale 1 = Binary */
  U8X8_CA(0x39, 0x36),            	/* (16) Gray Scale Pattern Set, Pattern */
  U8X8_CA(0x2B, 0x07),				/* (27) LCD Drive Mode Voltage Select, Parameter */
  U8X8_CA(0x81, 0x0a),		        /* (28) Electronic Volume, Parameter */
  U8X8_CA(0x5F, 0x00),		        /* (24) Built-in Oscillator Frequency, Parameter  */
  U8X8_C(0xAA|1),			        /* (23) Built-in OSC On */  
  U8X8_CA(0x25, 0x1f),		        /* (25) Power Control Set, Parameter  */
  U8X8_CA(0x8A, 0x00),		        /* (6) Start Line Setup, Parameter  */
  U8X8_CA(0xB1, 0x00),		        /* (7) Page Address Set, Parameter  */
  U8X8_CA(0x13, 0x00),		        /* (8) Column Address Set  */
  U8X8_C(0xAE|1),			        /* (1) Display ON/OFF */  
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

olikraus, 2021-04-25:
Thanks for the contribution
I have created constructors U8G2_S1D15721_240X64_1_4W_SW_SPI, U8G2_S1D15721_240X64_1_4W_HW_SPI, U8G2_S1D15721_240X64_F_4W_SW_SPI, U8G2_S1D15721_240X64_F_4W_HW_SPI etc
The SPI constructors are also added to each example.
I took over your modifications and fixed the offset problem (hopefully).
I have also created a new beta release 2.29.4. Maybe you can check whether the offset problem is fixed.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

loschwa, 2021-04-25:
@olikraus
Many thanks for the new U8g2 beta release and your excellent library in general. I just tested it and found that your release works flawlessly with the S1D15721! The offset is gone and the full display area can be addressed. Funny enough it appears that the display actually has 241x65 pixels since "u8g2.drawFrame(0,0,240,64);" produces a visible frame at the extreme border.

Btw.: the LCD-display only costs 0.95€ at https://www.pollin.de/p/lcd-varitronix-cog-vlgem1277-01-240x64-pixel-121713. I have designed a PCB for connecting to the flexprint cable for which I can provide the Eagle files if of interest.

olikraus, 2021-04-25:
Thanks for the feedback. I am happy that the display works.

And yes, I also saw this "schnäppchen" at pollin.de. Hard to believe...

You display unfortunately still has a width of only 240 pixel, because the 3rd argument for drawFrame is the width of the frame (not the right corner):
`void U8G2::drawFrame(u8g2_uint_t x, u8g2_uint_t y, u8g2_uint_t w, u8g2_uint_t h)`
https://github.com/olikraus/u8g2/wiki/u8g2reference#drawframe

loschwa, 2021-04-25:
Hoppla, das war mir entgangen. Danke für den Hinweis!

olikraus, 2021-04-25:
Kein Ding 😃

closed 2021-09-02 by olikraus

Jason2866, 2022-12-24:
@loschwa The display is still available from Pollin :-)
Can you provide the Eagle Files in your Github. Are there any external components needed? Or is just connecting to an MCU enough?

olikraus, 2022-12-24:
At least several 1uF caps are required according to the pin description.