olikraus/u8g2 #1475: HW SPI does not control MOSI and SCLK

loschwa, 2021-04-27:
Having used u8g2 in SW_SPI-mode I found that the MCU looses control of a I2C-RTC following  calling "u8g2.begin();". A similar problem has been reported other users before. In order to make sure that the problem is not related to the SW_SPI- mode. I used the constructor for HW_SPI:  

U8G2_S1D15721_240X64_F_4W_HW_SPI u8g2(U8G2_R0, cs, dc, res); //full framebuffer, size = 2080 bytes

Before doing this, I made sure that the HW_SPI-mode is working on the used MCU (WEMOS LOLIN32) and LCD-Display by using  SPI.h and the default MOSI, SCLK and CS Pins. The communication with the LCD via HW-SPI worked fine using a simple test program. The logic analyser output looks like this:
[image]

Following this I checked the logic analyser output of u8g2 in SW_SPI-mode:
[image]

The logic analyser output of u8g2 in HW_SPI-mode shows that the MOSI and SCLK lines are not toggled at all:
[image]

The relevant arduino code looks like this:

#include <Arduino.h>
#include <U8g2lib.h>
#include <TimeLib.h>
#include <Wire.h>
#include <SPI.h>

#define sclk 18
#define mosi 23
#define cs 5
#define dc 16
#define res 2

#define PCFaddress 0x50

int n;

//U8G2_S1D15721_240X64_F_4W_SW_SPI u8g2(U8G2_R0, sclk, mosi, cs, dc, res); //full framebuffer, size = 2080 bytes
U8G2_S1D15721_240X64_F_4W_HW_SPI u8g2(U8G2_R0, cs, dc, res); //full framebuffer, size = 2080 bytes

void setup(void) {

  pinMode(cs, OUTPUT);
  digitalWrite(cs,HIGH);
  pinMode(dc, OUTPUT);
  pinMode(res, OUTPUT);
  digitalWrite(res,HIGH);

  u8g2.begin();
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_fur42_tr);
  u8g2.sendBuffer(); 
}

void loop(void) {

  u8g2.clearBuffer();
  u8g2.setCursor(8, 52);
  u8g2.print("Hello!"); 
  u8g2.sendBuffer();
  
  n++;
  Serial.println(n);

  delay(1000);  
}

I do not believe that the problem is related to the display configuration file "u8x8_d_s1d15721.c" as the SW_SPI works without any problem. The SPI-speed and mode is also correct. Could the problem be related to the fact that the s1d15721 does not perform any handshaking with the host (MISO is not connected)?

loschwa, 2021-04-28:
Further tests show that if using a different constructor i.e. "U8G2_ST7565_64128N_1_4W_HW_SPI u8g2(U8G2_R0, cs, dc, res)" the SPI transfer continues without problem after the display init sequence. My assumption now is that the problem is related to the fact that the EPSON Controllers require during the init sequence commands with **two** arguments. All other controllers for which I checked the configuration files (e.g. u8x8_d_st7565.c), don't use commands with two arguments.

The funny thing is that u8g2 works in the SW-SPI configuration. The sequence:

// Sends Commands (3 Byte).
static void _lcd_cmd3(uint8_t cmd, uint8_t arg1, uint8_t arg2) {
  digitalWrite(PIN_CS,LOW);
  digitalWrite(PIN_A0,LOW);
  SPI.transfer(cmd);
  digitalWrite(PIN_A0,HIGH);
  SPI.transfer(arg1);
  SPI.transfer(arg2);
  digitalWrite(PIN_CS,HIGH);
}

should be the same, whether it is transfered by bit-banging or SPI HW.

loschwa, 2021-05-06:
In contrast to what I had assumed initially, further tests have shown that the problem is not related to the utilisation of the HW-SPI but rather to the complexity of the called function and the fact that the function is placed as angument to the 'u8g2.print' statement or not. If a String variable holds the result of a function then 'u8g2.print(stringvariable);' works apparently without problem. However, if the function is placed as an argument to the print-statement then artifacts appear on the display. The Heap and Stack sizes seem not to be related to the problem since they are ~370k and ~6.8k respectively.

Above has been tested with Buffer Size 1 or 2. Full frame buffer mode only works, if no external function calls are made. For instance calling 'ESP.getFreeHeap()' or just 'Serial.print' crashes the MCU right after booting.

I realise that the problem I am experiencing is possibly related to the HW configuration I am using (ESP32 and EPSON S1D15721).

closed 2021-05-06 by loschwa

olikraus, 2021-05-07:
Hmmm... Ok....