olikraus/u8g2 #1483: ST7920 Menu using u8g2 library

AnitaUniqa, 2021-05-02:
Hi everyone,
I would like to create a little menu with this library. I not found an example to do that, so i will try to create something for me and other.
I read the doc, but something goes wrong... can you help me?

- The part in the Loop, shows only the Selection List but not the functions.

`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
#define sampling 5
int kadarGula;
double Y1,Y2;
float V;

uint8_t current_selection = 0;
uint8_t response = 0;

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, 13, 11, 10, U8X8_PIN_NONE); //inisialisasi pin LCD

const char *menu1 =
  "Larutan Gula Non-Perisa\n"
  "Larutan Berperisa";
  
void setup(void) {
  u8g2.begin(/*Select=*/12, A1,A2,/*Up=*/9, /*Down=*/7); // Pin Button
  u8g2.firstPage();
  do {
    awal(); //tampilan WM
  } while ( u8g2.nextPage() );
  delay(500);
}

void awal(void) //Program Welcome Message
{
  u8g2.drawFrame(0, 0, 128, 63); //INISIALISASI FRAME
  u8g2.setFont(u8g2_font_glasstown_nbp_tf); //INISIALISASI FONT
  u8g2.drawStr(25, 15, "SELAMAT DATANG");
  u8g2.drawStr(35, 35, "Batasi Gulamu");
  u8g2.drawStr(15, 50, "Sebelum Terlambat ^^");
}

void draw_nonperisa() {
  u8g2.drawFrame(0, 0, 73, 32); //INISIALISASI FRAME
  u8g2.drawFrame(73, 0, 55, 32);
  u8g2.setFont(u8g2_font_glasstown_nbp_tf); //INISIALISASI FONT

  //KADAR GULA
  u8g2.drawStr(13, 13, "Kadar Gula"); //samping*tinggi
  for (int i = 0; i < sampling; i++) {
    kadarGula += analogRead(A0);
  }
  kadarGula = kadarGula / sampling;
  V = (kadarGula * 5.0) / 1023; //pin sensor
  Y1 = (0.0466 * V) + 0.6329; //rumus regresi NON PERISA
  u8g2.setCursor(5, 27);
  u8g2.print(Y1,2);
  u8g2.drawStr(28, 27, "Gram/dL"); //samping*tinggi
  Serial.println(Y1);

  //STATUS KADAR GULA
  u8g2.drawStr(86, 13, "Status"); //samping*tinggi
  if (Y1 > 0.79)
  {
    u8g2.drawStr(85, 27, "Tinggi"); //samping*tinggi
  }

  if (Y1 < 0.78)
  {
    u8g2.drawStr(85, 27, "Rendah"); //samping*tinggi
  }

  if (Y1 >= 0.78 && Y1 <= 0.79)
  {
    u8g2.drawStr(85, 27, "Sedang"); //samping*tinggi
  }

  //PESAN (EARLY WARNING SYSTEM)
  if (Y1 > 0.79)
  {
    u8g2.drawStr(40, 45, "WARNING !");
    u8g2.drawStr(16, 58, "KURANGI PORSI GULA");
  }

  if (Y1 < 0.78)
  {
    u8g2.drawStr(50, 45, "AMAN"); //
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }

  if (Y1 >= 0.78 && Y1 <=0.79)
  {
    u8g2.drawStr(35, 45, "BATAS AMAN");
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }
}

void draw_perisa() {
  u8g2.drawFrame(0, 0, 73, 32); //INISIALISASI FRAME
  u8g2.drawFrame(73, 0, 55, 32);
  u8g2.setFont(u8g2_font_glasstown_nbp_tf); //INISIALISASI FONT

  //KADAR GULA
  u8g2.drawStr(13, 13, "Kadar Gula"); //samping*tinggi
  for (int i = 0; i < sampling; i++) {
    kadarGula += analogRead(A0);
  }
  kadarGula = kadarGula / sampling;
  V = (kadarGula * 5.0) / 1023; //pin sensor
  Y2 = (0.0568 * V) + 0.7098; //Rumus regresi PERISA
  //dtostrf(Y1, 7, 2, kadar1);
  //u8g2.drawStr(0, 27, kadar1); //samping*tinggi Output sensor
  u8g2.setCursor(5, 27);
  u8g2.print(Y2,2);
  u8g2.drawStr(28, 27, "Gram/dL"); //samping*tinggi
  Serial.println(Y2);

  //STATUS KADAR GULA
  u8g2.drawStr(86, 13, "Status"); //samping*tinggi
  if (Y2 > 0.79)
  {
    u8g2.drawStr(85, 27, "Tinggi"); //samping*tinggi
  }

  if (Y2 < 0.78)
  {
    u8g2.drawStr(85, 27, "Rendah"); //samping*tinggi
  }

  if (Y2 >= 0.78 && Y2 <= 0.79)
  {
    u8g2.drawStr(85, 27, "Sedang"); //samping*tinggi
  }

  //PESAN (EARLY WARNING SYSTEM)
  if (Y2 > 0.79)
  {
    u8g2.drawStr(40, 45, "WARNING !");
    u8g2.drawStr(16, 58, "KURANGI PORSI GULA");
  }

  if (Y2 < 0.78)
  {
    u8g2.drawStr(50, 45, "AMAN"); //
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }

  if (Y2 >= 0.78 && Y2 <=0.79)
  {
    u8g2.drawStr(35, 45, "BATAS AMAN");
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }
}

void loop(void) {
  current_selection = u8g2.userInterfaceSelectionList(
                        "Mode Pengukuran",
                        current_selection,
                        menu1);
  delay(15); //debounce
  response = u8g2.userInterfaceMessage(
               "Pilihan:\n",
               u8x8_GetStringLineStart(current_selection-1, menu1),
               "",
               " OK ");

  if (response == 1) {         // NON-Perisa
    draw_nonperisa();
  }

  if (response == 2) {         // Perisa
    draw_perisa();
  }
}
`

olikraus, 2021-05-02:
u8g2.sendBuffer() is missing (https://github.com/olikraus/u8g2/wiki/u8g2reference#sendbuffer).

AnitaUniqa, 2021-05-03:
I did some changes in my code, it's much more scalable now.
The menu do:

- When I click the "OK" menu in an option list

1. Shows the process in the draw_perisa / draw_non perisa function, but only briefly and returns to the menu.
2. Overlap view between message box and process view

How to display a processes with a one-click menu and do not overlap? Did I use something function in a wrong way?
`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
#define sampling 5
int kadarGula;
double Y1,Y2;
float V;
//char kadar[5]; //konversi nilai kadargula menjadi karakter
uint8_t current_selection = 0;
uint8_t response = 0;

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, 13, 11, 10, U8X8_PIN_NONE); //inisialisasi pin LCD

const char *menu1 =
  "Larutan Gula Non-Perisa\n"
  "Larutan Berperisa";
  
void setup(void) {
  u8g2.begin(/*Select=*/12, A1,A2,/*Up=*/9, /*Down=*/7); // Pin Button
  /*u8g2.setFont(u8g2_font_6x12_tr);
    drawText(15, 40, "Welcome");*/
  u8g2.firstPage();
  do {
    awal(); //tampilan WM
  } while ( u8g2.nextPage() );
  delay(500);
}

/*void drawText(byte X, byte Y, const char *text) {
  u8g2.firstPage();
  do {
    u8g2.drawStr(X, Y, text);
  } while ( u8g2.nextPage() );
  delay(1000);
  }*/

void awal(void) //Program Welcome Message
{
  u8g2.drawFrame(0, 0, 128, 63); //INISIALISASI FRAME
  u8g2.setFont(u8g2_font_glasstown_nbp_tf); //INISIALISASI FONT
  u8g2.drawStr(25, 15, "SELAMAT DATANG");
  u8g2.drawStr(35, 35, "Batasi Gulamu");
  u8g2.drawStr(15, 50, "Sebelum Terlambat ^^");
}

void draw_nonperisa() {
  u8g2.drawFrame(0, 0, 73, 32); //INISIALISASI FRAME
  u8g2.drawFrame(73, 0, 55, 32);
  u8g2.setFont(u8g2_font_glasstown_nbp_tf); //INISIALISASI FONT

  //KADAR GULA
  u8g2.drawStr(13, 13, "Kadar Gula"); //samping*tinggi
  for (int i = 0; i < sampling; i++) {
    kadarGula += analogRead(A0);
  }
  kadarGula = kadarGula / sampling;
  V = (kadarGula * 5.0) / 1023; //pin sensor
  Y1 = (0.0466 * V) + 0.6329; //rumus regresi NON PERISA
  //dtostrf(Y1, 7, 2, kadar1);
  //u8g2.drawStr(0, 27, kadar1); //samping*tinggi Output sensor
  u8g2.setCursor(5, 27);
  u8g2.print(Y1,2);
  u8g2.drawStr(28, 27, "Gram/dL"); //samping*tinggi
  Serial.println(Y1);

  //STATUS KADAR GULA
  u8g2.drawStr(86, 13, "Status"); //samping*tinggi
  if (Y1 > 0.79)
  {
    u8g2.drawStr(85, 27, "Tinggi"); //samping*tinggi
  }

  if (Y1 < 0.78)
  {
    u8g2.drawStr(85, 27, "Rendah"); //samping*tinggi
  }

  if (Y1 >= 0.78 && Y1 <= 0.79)
  {
    u8g2.drawStr(85, 27, "Sedang"); //samping*tinggi
  }

  //PESAN (EARLY WARNING SYSTEM)
  if (Y1 > 0.79)
  {
    u8g2.drawStr(40, 45, "WARNING !");
    u8g2.drawStr(16, 58, "KURANGI PORSI GULA");
  }

  if (Y1 < 0.78)
  {
    u8g2.drawStr(50, 45, "AMAN"); //
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }

  if (Y1 >= 0.78 && Y1 <=0.79)
  {
    u8g2.drawStr(35, 45, "BATAS AMAN");
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }
}

void draw_perisa() {
  u8g2.drawFrame(0, 0, 73, 32); //INISIALISASI FRAME
  u8g2.drawFrame(73, 0, 55, 32);
  u8g2.setFont(u8g2_font_glasstown_nbp_tf); //INISIALISASI FONT

  //KADAR GULA
  u8g2.drawStr(13, 13, "Kadar Gula"); //samping*tinggi
  for (int i = 0; i < sampling; i++) {
    kadarGula += analogRead(A0);
  }
  kadarGula = kadarGula / sampling;
  V = (kadarGula * 5.0) / 1023; //pin sensor
  Y2 = (0.0568 * V) + 0.7098; //Rumus regresi PERISA
  //dtostrf(Y1, 7, 2, kadar1);
  //u8g2.drawStr(0, 27, kadar1); //samping*tinggi Output sensor
  u8g2.setCursor(5, 27);
  u8g2.print(Y2,2);
  u8g2.drawStr(28, 27, "Gram/dL"); //samping*tinggi
  Serial.println(Y2);

  //STATUS KADAR GULA
  u8g2.drawStr(86, 13, "Status"); //samping*tinggi
  if (Y2 > 0.79)
  {
    u8g2.drawStr(85, 27, "Tinggi"); //samping*tinggi
  }

  if (Y2 < 0.78)
  {
    u8g2.drawStr(85, 27, "Rendah"); //samping*tinggi
  }

  if (Y2 >= 0.78 && Y2 <= 0.79)
  {
    u8g2.drawStr(85, 27, "Sedang"); //samping*tinggi
  }

  //PESAN (EARLY WARNING SYSTEM)
  if (Y2 > 0.79)
  {
    u8g2.drawStr(40, 45, "WARNING !");
    u8g2.drawStr(16, 58, "KURANGI PORSI GULA");
  }

  if (Y2 < 0.78)
  {
    u8g2.drawStr(50, 45, "AMAN"); //
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }

  if (Y2 >= 0.78 && Y2 <=0.79)
  {
    u8g2.drawStr(35, 45, "BATAS AMAN");
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }
}

void loop(void) {
  u8g2.clearBuffer();
  current_selection = u8g2.userInterfaceSelectionList(
                        "Mode Pengukuran",
                        current_selection,
                        menu1);
  delay(15); //debounce
  response = u8g2.userInterfaceMessage(
               "Pilihan:\n",
               u8x8_GetStringLineStart(current_selection-1, menu1),
               "",
               " OK ");

  if (response == 1) {         // NON-Perisa
    draw_nonperisa();
  }

  if (response == 2) {         // Perisa
    draw_perisa();
  }
  u8g2.sendBuffer();
  delay(1000);
}`

AnitaUniqa, 2021-05-03:
Thx @olikraus 

I did some changes in my code, it's much more scalable now.
The menu do:

- When I click the "OK" menu in an option list

1. Shows the process in the draw_perisa / draw_non perisa function, but only briefly and returns to the menu.
2. Overlap view between message box and process view

How to display a processes with a one-click menu and do not overlap? Did I use something function in a wrong way?

`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
#define sampling 5
int kadarGula;
double Y1,Y2;
float V;
uint8_t current_selection = 0;
uint8_t response = 0;

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, 13, 11, 10, U8X8_PIN_NONE); //inisialisasi pin LCD

const char *menu1 =
  "Larutan Gula Non-Perisa\n"
  "Larutan Berperisa";
  
void setup(void) {
  u8g2.begin(/*Select=*/12, A1,A2,/*Up=*/9, /*Down=*/7); // Pin Button
  u8g2.firstPage();
  do {
    awal(); //tampilan WM
  } while ( u8g2.nextPage() );
  delay(500);

void awal(void) //Program Welcome Message
{
  u8g2.drawFrame(0, 0, 128, 63); //INISIALISASI FRAME
  u8g2.setFont(u8g2_font_glasstown_nbp_tf); //INISIALISASI FONT
  u8g2.drawStr(25, 15, "SELAMAT DATANG");
  u8g2.drawStr(35, 35, "Batasi Gulamu");
  u8g2.drawStr(15, 50, "Sebelum Terlambat ^^");
}

void draw_nonperisa() {
  u8g2.drawFrame(0, 0, 73, 32); //INISIALISASI FRAME
  u8g2.drawFrame(73, 0, 55, 32);
  u8g2.setFont(u8g2_font_glasstown_nbp_tf); //INISIALISASI FONT

  //KADAR GULA
  u8g2.drawStr(13, 13, "Kadar Gula"); //samping*tinggi
  for (int i = 0; i < sampling; i++) {
    kadarGula += analogRead(A0);
  }
  kadarGula = kadarGula / sampling;
  V = (kadarGula * 5.0) / 1023; //pin sensor
  Y1 = (0.0466 * V) + 0.6329; //rumus regresi NON PERISA
  //dtostrf(Y1, 7, 2, kadar1);
  //u8g2.drawStr(0, 27, kadar1); //samping*tinggi Output sensor
  u8g2.setCursor(5, 27);
  u8g2.print(Y1,2);
  u8g2.drawStr(28, 27, "Gram/dL"); //samping*tinggi
  Serial.println(Y1);

  //STATUS KADAR GULA
  u8g2.drawStr(86, 13, "Status"); //samping*tinggi
  if (Y1 > 0.79)
  {
    u8g2.drawStr(85, 27, "Tinggi"); //samping*tinggi
  }

  if (Y1 < 0.78)
  {
    u8g2.drawStr(85, 27, "Rendah"); //samping*tinggi
  }

  if (Y1 >= 0.78 && Y1 <= 0.79)
  {
    u8g2.drawStr(85, 27, "Sedang"); //samping*tinggi
  }

  //PESAN (EARLY WARNING SYSTEM)
  if (Y1 > 0.79)
  {
    u8g2.drawStr(40, 45, "WARNING !");
    u8g2.drawStr(16, 58, "KURANGI PORSI GULA");
  }

  if (Y1 < 0.78)
  {
    u8g2.drawStr(50, 45, "AMAN"); //
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }

  if (Y1 >= 0.78 && Y1 <=0.79)
  {
    u8g2.drawStr(35, 45, "BATAS AMAN");
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }
}

void draw_perisa() {
  u8g2.drawFrame(0, 0, 73, 32); //INISIALISASI FRAME
  u8g2.drawFrame(73, 0, 55, 32);
  u8g2.setFont(u8g2_font_glasstown_nbp_tf); //INISIALISASI FONT

  //KADAR GULA
  u8g2.drawStr(13, 13, "Kadar Gula"); //samping*tinggi
  for (int i = 0; i < sampling; i++) {
    kadarGula += analogRead(A0);
  }
  kadarGula = kadarGula / sampling;
  V = (kadarGula * 5.0) / 1023; //pin sensor
  Y2 = (0.0568 * V) + 0.7098; //Rumus regresi PERISA
  //dtostrf(Y1, 7, 2, kadar1);
  //u8g2.drawStr(0, 27, kadar1); //samping*tinggi Output sensor
  u8g2.setCursor(5, 27);
  u8g2.print(Y2,2);
  u8g2.drawStr(28, 27, "Gram/dL"); //samping*tinggi
  Serial.println(Y2);

  //STATUS KADAR GULA
  u8g2.drawStr(86, 13, "Status"); //samping*tinggi
  if (Y2 > 0.79)
  {
    u8g2.drawStr(85, 27, "Tinggi"); //samping*tinggi
  }

  if (Y2 < 0.78)
  {
    u8g2.drawStr(85, 27, "Rendah"); //samping*tinggi
  }

  if (Y2 >= 0.78 && Y2 <= 0.79)
  {
    u8g2.drawStr(85, 27, "Sedang"); //samping*tinggi
  }

  //PESAN (EARLY WARNING SYSTEM)
  if (Y2 > 0.79)
  {
    u8g2.drawStr(40, 45, "WARNING !");
    u8g2.drawStr(16, 58, "KURANGI PORSI GULA");
  }

  if (Y2 < 0.78)
  {
    u8g2.drawStr(50, 45, "AMAN"); //
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }

  if (Y2 >= 0.78 && Y2 <=0.79)
  {
    u8g2.drawStr(35, 45, "BATAS AMAN");
    u8g2.drawStr(18, 58, "Silahkan dikonsumsi :)");
  }
}

void loop(void) {
  u8g2.clearBuffer();
  current_selection = u8g2.userInterfaceSelectionList(
                        "Mode Pengukuran",
                        current_selection,
                        menu1);
  delay(15); //debounce
  response = u8g2.userInterfaceMessage(
               "Pilihan:\n",
               u8x8_GetStringLineStart(current_selection-1, menu1),
               "",
               " OK ");

  if (response == 1) {         // NON-Perisa
    draw_nonperisa();
  }

  if (response == 2) {         // Perisa
    draw_perisa();
  }
  u8g2.sendBuffer();
  delay(1000);
}`

chandanchan0086, 2021-05-07:
Hi @AnitaUniqa I'm trying to do the same menu option selection using button, but im using Esp-32 and ESP-IDF frame work.
i didn't found any examples .... can you please help me out with this?

olikraus, 2021-05-09:
Not sure whether i can help further here. U8g2 only deals with low level display access. Menu libs are usually build on top of u8g2 (like https://github.com/neu-rah/ArduinoMenu)

olikraus, 2022-03-04:
the u8g2 button detection is more like a demo addon... It did work in u8g2 demo code.

closed 2022-03-04 by olikraus