olikraus/u8g2 #1484: Easy way to fix some font artifacts

regdomain, 2021-05-04:
Thanks for your huge work over these years, I am very much impressed!
I am trying to find an easy way to fix some built-in font artifacts. What do I mean? For example logisoso font: digit 3 is located one pixel lower than the other digits, digit 2 has some extra pixels set at the beak, colon is missing some pixels. There are others also.
To save time to create a new font with the third party tools I believe it would be much easier just to edit the existing fonts. However I cannot find an easy way to convert c font format to bdf or any other editable format in order to convert it back to c with bdfconv after fixing the artifacts. There's a nice tool https://andrzejo.github.io/demo/u8x8-font-editor.html which could absolutely help me but it only supports u8x8, not u8g2 font format.
Are there ALL source bdf files published (I only found SOME on github)? Or is there a c2bdf converter? Or am I missing something in general to solve this esthetic "problem"?

olikraus, 2021-05-04:
A lot of good questions.

> Are there ALL source bdf files published (I only found SOME on github)? 

U8g2/u8x8 fonts are derived from two different sources: .ttf and .bdf files. The workflows are:
.ttf -- otf2bdf --> tmp.bdf -- bdfconv --> .c
and
.bdf -- bdfconv --> .c

.ttf fonts are located here: https://github.com/olikraus/u8g2/tree/master/tools/font/ttf
.bdf fonts are located here: https://github.com/olikraus/u8g2/tree/master/tools/font/bdf

To make things more complicated: .ttf (truetype font) might include a pixel font or a scalable font. From outside there is no difference and also the conversion process using otf2bdf is the same, but scalable fonts will show the mentioned artifacts.

In other words, the artifacts you have observed will not appear in cases where the source is a .bdf file or with a .ttf file, which includes a pixel font. 
Lets have a look at the logisoso font. This is indeed a scalable font, which is located here: https://github.com/olikraus/u8g2/blob/master/tools/font/ttf/Logisoso.ttf

[image]

Indeed "3" is little bit lower than "2", however the "8" seems to be correct.
Here is a picture of the original ttf file, magnified with gimp:

[image]

logisoso is a ttf font which is intended to be used on a full color, RGB screen: It creates a graylevel image of each char and as you can see, there is a small gray line below the baseline of the "3" which is not there for "8".
During the conversion to a black white image by otf2bdf, this gray line is converted to black which then appear as black pixel below the baseline. 

From my point of view, this has to be fixed in the .ttf file itself. I do not know how and why "3" differs from "8", but I assume the construction of "8" has to be taken over to "3"

> Or is there a c2bdf converter?

No, all sources are available.

> Or am I missing something in general to solve this esthetic "problem"?

The biggest problem is the rasterization in general. If a ttf font is not designed to look good on a black and white target device, the the probability for the mentioned artifacts is very high.

So, how could this be solved. You probably need to use fontforge: Load the ttf font into fontforge, change the ttf so that it does not generate the above mentioned gray line below baseline, save the ttf, apply otf2bdf and then bdfconv.

regdomain, 2021-05-05:
Dear Oliver, thank you again for your detailed answer and comprehensive explanation. The root cause is rasterization, it is clear. And a reverse process I was searching for makes no sense as all the files are there. The normal conversion according to your workflow but with fixing artifacts in bdf will do the job.

regdomain, 2021-05-09:
I am wondering where I can find the source files for .c fonts located in https://github.com/olikraus/u8g2/tree/master/tools/font/build/single_font_files
Thanks.

olikraus, 2021-05-09:
The .c files are the result of the u8g2 font building tool: https://github.com/olikraus/u8g2/blob/master/tools/font/build/build.c
However, the sources are still the mentioned locations:

.ttf fonts are located here: https://github.com/olikraus/u8g2/tree/master/tools/font/ttf
.bdf fonts are located here: https://github.com/olikraus/u8g2/tree/master/tools/font/bdf

Let me use an example. Assume the following file:
https://github.com/olikraus/u8g2/blob/master/tools/font/build/single_font_files/u8g2_font_blipfest_07_tn.c

It is constructed by this line:
https://github.com/olikraus/u8g2/blob/19929687e2f55d1296bf5ba03a0dc73a27947c7a/tools/font/build/build.c#L970

According to this line, the source file is here:
https://github.com/olikraus/u8g2/blob/master/tools/font/ttf/blipfest_07.ttf

The _tn name postfix was caused by the BM_T and MM_N optiona in the same line.

regdomain, 2021-05-09:
Ah, now I see, several fonts are generated from the same source file!
By the way I have created a couple of Russian fonts 8 pixel height which are apparently missing now.
Would it be reasonable to share them here?

olikraus, 2021-05-09:
You could add the source files here. But I can not add those files to u8g2 die to GitHub bandwidth restrictions for the documentation page.

closed 2022-03-04 by olikraus