olikraus/u8g2 #1486: Why #include <Arduino.h> explicitly?

AndrewBCN, 2021-05-06:
Hello,
I have encountered a compilation error when compiling the various u8x8 example sketches for STM32duino. When I removed the explicit
#include <Arduino.h>
the compilation error disappeared, and the examples ran fine.
Since <Arduino.h> is already automatically included by the Arduino build system, I was wondering why it is included explicitly in the u8g2/u8x8 examples.
Isn't it redundant to include it explicitly? Or does it serve a specific purpose?

olikraus, 2021-05-06:
Where exactly did you remove the #include <Arduino.h>? From U8x8lib.h? 
What kind of error messages did you get?

AndrewBCN, 2021-05-06:
I apologize, I should have provided more information. Here are more details:
In the u8x8 example sketch Hello World, compiled for an STM32F411CEU6 Black Pill in Arduino IDE 1.8.13 with STM32 core release 2.0.0 installed:

1) This code compiles fine:
//#if !defined(STM32_CORE_VERSION) || (STM32_CORE_VERSION  < 0x02000000)
//#error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x02000000"
//#endif

#include <Arduino.h>
#include <U8x8lib.h>

#include <Wire.h>

U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE); 	      

void setup(void)
{
  u8x8.begin();
  u8x8.setPowerSave(0);
}

void loop(void)
{
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  u8x8.drawString(0,1,"Hello World!");
  delay(2000);
}

2) If I comment out the #include <Arduino.h>, it also compiles and runs fine. Apparently the Arduino build system automatically includes <Arduino.h> when preprocessing the .ino file.
//#if !defined(STM32_CORE_VERSION) || (STM32_CORE_VERSION  < 0x02000000)
//#error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x02000000"
//#endif

// #include <Arduino.h>
#include <U8x8lib.h>

#include <Wire.h>
...

3) If I add the STM32 core version test, and leave the #include <Arduino.h> commented out, it also compiles fine:
#if !defined(STM32_CORE_VERSION) || (STM32_CORE_VERSION  < 0x02000000)
#error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x02000000"
#endif

// #include <Arduino.h>
#include <U8x8lib.h>

#include <Wire.h>
...

4) And if I try to test the core version and uncomment the #include <Arduino.h>, this triggers the error in the test during compilation:
#if !defined(STM32_CORE_VERSION) || (STM32_CORE_VERSION  < 0x02000000)
#error "Due to API change, this sketch is compatible with STM32_CORE_VERSION  >= 0x02000000"
#endif

#include <Arduino.h>
#include <U8x8lib.h>

#include <Wire.h>
...

I did not touch U8x8lib.h or any other U8g2/u8x8 library file.

olikraus, 2021-05-06:
Thanks for the details.
Actually I do not know what is wrong here, but it looks like you have several working option, right? So it's not a real problem, more like a strange behaviour, correct?

AndrewBCN, 2021-05-06:
Indeed it is not at all a serious problem, as the simple workaround - removing the redundant #include <Arduino.h> in the U8g2/u8x8 example sketches - works fine.
I am closing the issue here, and I have opened an issue in the Arduino github for this, perhaps they can explain or find what the problem is exactly.

closed 2021-05-06 by AndrewBCN