olikraus/u8g2 #1488: UC1601 128x64 Library

uuygun, 2021-05-07:
Hello, there are UC1601 128x32 codes in the library, but I am using UC1601 128x64 and there are no codes for this. And these codes are not working for my screen. What can I do about it?

olikraus, 2021-05-09:
The UC1601 is almost compatible with many other controllers, especially the ST7565. 

https://github.com/olikraus/u8g2/blob/master/doc/controller_cmds.txt

My suggestion is to test the constructors with the ST7565 128x64 controller.

olikraus, 2021-05-10:
Did you add the required 1uF caps for the charge pump? They should be as close as possible to the flex cable connector.

uuygun, 2021-05-10:
Yes, i added the capacitor pretty close.

olikraus, 2021-05-10:
Maybe you could still check with some other constructors.

closed 2022-03-04 by olikraus