olikraus/u8g2 #1489: Blank screen on RA8835 320x240 protocol 6800.

Finside, 2021-05-07:
I would really appreciate any help with RA8835 320x240 display. Didn't noticed this RA8835 chip and didn't check if it works with U8Glib2. Now my Boss will kill all 3 of us because of this display! Any help is highly appreciated.

The only working setup is
U8G2_RA8835_NHD_240X128_1_6800
but on 320x240 it works incorrectly.

Now we are trying to analyze both
U8G2_RA8835_320X240_1_6800
and
U8G2_RA8835_NHD_240X128_1_6800
to find differences and fix the problem.

[image]

olikraus, 2021-05-08:
Did you enable 16 bit mode for u8g2? See FAQ.

olikraus, 2021-05-09:
Which constructor is visible on your picture? Is this the U8G2_RA8835_320X240_1_6800?

ichetiva, 2021-12-24:
> I would really appreciate any help with RA8835 320x240 display with 8 bit parallel interface. I connect it to Arduino Mega 2560.
> 
> U8G2_16BIT uncommented and checked.
> 
> Installation:
> 
> 1. Arduino IDE 1.8.12 from ZIP manually unpacked into new folder.
> 2. Into Arduino/libraries manually added folder with U8Glib2.
> 
> Didn't noticed this RA8835 chip and didn't check if it works with U8Glib2. Now my Boss will kill all 3 of us because of this display! Any help is highly appreciated.
> 
> The only working setup is U8G2_RA8835_NHD_240X128_1_6800 but on 320x240 it works incorrectly.
> 
> Now we are trying to analyze both U8G2_RA8835_320X240_1_6800 and U8G2_RA8835_NHD_240X128_1_6800 to find differences and fix the problem.
> 
> [image]

Did you manage to display at least something on this screen?

Finside, 2021-12-25:
Did you manage to display at least something on this screen?
//
Sorry, I haven't for the question. As you can see, there is some text on the screen.

ichetiva, 2021-12-25:
> Did you manage to display at least something on this screen? // Sorry, I haven't for the question. As you can see, there is some text on the screen.

In image display 320x240?

Finside, 2021-12-25:
Yes, 320x240 works fine, but very slow (2.8 sec) on Arduino Mega 2560 with 8 bit parallel interface.

[image]

Finside, 2021-12-25:
Image 
[image]

Finside, 2021-12-25:
[image]
25 FPS 480p Video to .GIF

ichetiva, 2021-12-26:
> [image] 25 FPS 480p Video to .GIF

А Вы не знаете что делать если текст выводится зеркально?

Finside, 2021-12-26:
- What to do if text is mirrored?

- I think we need to compare:
1. All connection lines Microcontroller -> Display
2. Display controller RA8835 
3. Library sttings (pins).

It is not possible to answer why text is mirrored without any additional info.

ichetiva, 2021-12-27:
https://user-images.githubusercontent.com/81914195/147498471-47893670-740f-407f-b9df-d4d168a2583c.MOV

Watch this video, here is the example of "PrintUTF8"

ichetiva, 2021-12-27:
What do you think is the matter?

olikraus, 2022-03-04:
hmm no further feedback, closing...

closed 2022-03-04 by olikraus