olikraus/u8g2 #1490: Cannot display u8g2

AnitaUniqa, 2021-05-08:
Hi everyone,
I would like to create a little menu option selection with this library. I not found an example to do that, so i will try to create something for me and other.

I read an example about selectionlist, but something goes wrong. The LCD can't display anything... can you help me?

`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
#define sampling 5
int kadarGula;
double Y1, Y2;
float V;
uint8_t current_selection = 0;
uint8_t response = 0;

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, 13, 11, 10, U8X8_PIN_NONE); //inisialisasi pin LCD

#define btnUp   9
#define btnOk   12
#define btnDown 7
#define btnBack 6

boolean statusBtnUp   = false;
boolean statusBtnOk   = false;
boolean statusBtnDown = false;
boolean statusBtnBack = false;

boolean statusAkhirBtnUp   = false;
boolean statusAkhirBtnOk   = false;
boolean statusAkhirBtnDown = false;
boolean statusAkhirBtnBack = false;

boolean UP   = false;
boolean OK   = false;
boolean DOWN = false;
boolean BACK = false;

int halaman  = 1;
int menuItem = 1;

void setup() {
  Serial.begin(9600);
  u8g2.begin();
  u8g2.clearBuffer();

  pinMode(btnUp,   INPUT_PULLUP);
  pinMode(btnOk,   INPUT_PULLUP);
  pinMode(btnDown, INPUT_PULLUP);
  pinMode(btnBack, INPUT_PULLUP);
}

void loop() {
  tampil();

  statusBtnUp   = digitalRead(btnUp);
  statusBtnOk   = digitalRead(btnOk);
  statusBtnDown = digitalRead(btnDown);
  statusBtnBack = digitalRead(btnBack);

  saatUpDitekan();
  saatOkDitekan();
  saatDownDitekan();
  saatBackDitekan();

  //  untuk button up
  if (UP && halaman == 1) {
    UP = false;
    menuItem --;
    if (menuItem < 1)menuItem = 4;
  }
  //untuk button down
  if (DOWN && halaman == 1) {
    DOWN = false;
    menuItem ++;
    if (menuItem > 4)menuItem = 1;
  }

  //  untuk button ok
  if (OK) {
    OK = false;
    if (halaman == 1 && menuItem == 1) {
      halaman = 2;
    } else if (halaman == 1 && menuItem == 2) {
      halaman = 3;
    } else if (halaman == 1 && menuItem == 3) {
      halaman = 4;
    } else if (halaman == 1 && menuItem == 4) {
      halaman = 5;
    }
  }

  //  untuk button back
  if (BACK) {
    BACK = false;
    if (halaman == 2 || halaman == 3 || halaman == 4 || halaman == 5) {
      halaman = 1;
    }
  }
  Serial.println(menuItem);
  delay(100);
}

//--------------------------------------------------------------------------------
void saatUpDitekan() {
  if (statusBtnUp != statusAkhirBtnUp) {
    if (statusBtnUp == 0) {
      UP = true;
    }
    delay(50);
  }
  statusAkhirBtnUp = statusBtnUp;
}

void saatOkDitekan() {
  if (statusBtnOk != statusAkhirBtnOk) {
    if (statusBtnOk == 0) {
      OK = true;
    }
    delay(50);
  }
  statusAkhirBtnOk = statusBtnOk;
}

void saatDownDitekan() {
  if (statusBtnDown != statusAkhirBtnDown) {
    if (statusBtnDown == 0) {
      DOWN = true;
    }
    delay(50);
  }
  statusAkhirBtnDown = statusBtnDown;
}

void saatBackDitekan() {
  if (statusBtnBack != statusAkhirBtnBack) {
    if (statusBtnBack == 0) {
      BACK = true;
    }
    delay(50);
  }
  statusAkhirBtnBack = statusBtnBack;
}
//----------------------------------------------------------------------------

//semua yang tampil di lcd ada di fungsi ini
void tampil() {
  if (halaman == 1) {
    u8g2.clearBuffer();
    u8g2.setCursor(30, 0);
    u8g2.print("DAFTAR MENU");

    if (menuItem == 1) {
      u8g2.setCursor(5, 17);
      u8g2.print("> menuItem 1");
    } else {
      u8g2.setCursor(5, 17);
      u8g2.print("  menuItem 1");
    }

    if (menuItem == 2) {
      u8g2.setCursor(5, 27);
      u8g2.print("> menuItem 2");
    } else {
      u8g2.setCursor(5, 27);
      u8g2.print("  menuItem 2");
    }

    if (menuItem == 3) {
      u8g2.setCursor(5, 37);
      u8g2.print("> menuItem 3");
    } else {
      u8g2.setCursor(5, 37);
      u8g2.print("  menuItem 3");
    }

    if (menuItem == 4) {
      u8g2.setCursor(5, 47);
      u8g2.print("> menuItem 4");
    } else {
      u8g2.setCursor(5, 47);
      u8g2.print("  menuItem 4");
    }

  } else if (halaman == 2) {
    u8g2.clearBuffer();
    u8g2.setCursor(25, 25);
    u8g2.print("ImamID");
  } else if (halaman == 3) {
    u8g2.clearBuffer();
    u8g2.setCursor(10, 15);
    u8g2.print("SUBSCRIBE");
    u8g2.setCursor(30, 35);
    u8g2.print("ImamID");
  } else if (halaman == 4) {
    u8g2.clearBuffer();
    u8g2.setCursor(25, 25);
    u8g2.print("ini hal 4");
  } else if (halaman == 5) {
    u8g2.clearBuffer();
    u8g2.setCursor(25, 25);
    u8g2.print("ini hal 5");
  }

  u8g2.sendBuffer();
}`

olikraus, 2022-03-04:
hmmm guess, I missed this. Do the original u8g2 examples work with your display?

olikraus, 2023-02-22:
no further feedback, closing...

closed 2023-02-22 by olikraus