olikraus/u8g2 #1492: LCD Dot matrix with Toshiba T7932 and T7933

spi5498, 2021-05-12:
Hi!

I have this really nice LCD module that i pulled out of a old POS-terminal.
Could not find much on that particular module, however the controller consists of 3x T7932 chips, and one T7933 chip.
I take a shot in the dark and hope that this may be some "standard" already compatible with another driver?

What do you think? Datasheets: [https://www.alldatasheet.com/datasheet-pdf/pdf/31153/TOSHIBA/T7932.html](7932) & [https://pdf1.alldatasheet.com/datasheet-pdf/view/31154/TOSHIBA/T7933.html](7933)

spi5498, 2021-05-12:
Also, if you have a hunch that it could be compatible with something, ill happely try it and report back

spi5498, 2021-05-22:
Hello again,

I found this: https://forum.arduino.cc/t/openglcd-hd44102-to-t7932/413755 
So it seems that would be a good starting point, does U8g2 have a driver for HD44102 or equivalent?

olikraus, 2021-05-22:
Hmmm... never heard about hd44102 LCD. It is not supported :-(

olikraus, 2021-05-23:
I will try to buy and add support for such a display with HD44102, but it might take 2-3 months (such displays are not easy to get...)

spi5498, 2021-05-23:
That's great!
However, this week i will receive a FPC adapter for my module T7932/T7933. 
I'm no LCD-expert, but with a little assistance I will be more than happy to try and help out. 
Do you have any hunch on if there is a "base" module that might be good to use that is in the library right now when looking at the datasheet?

olikraus, 2021-05-24:
Maybe the ks0108 could be a starting point.

spi5498, 2021-05-25:
Thank you!

Will give that a shot when I get my adapter and report back!

olikraus, 2021-06-15:
Some progress here. I might get a LM324 display with the HD44102 controller.

spi5498, 2021-06-16:
Hi Oli!
Thank you very much for this update. 
In the last days I have received my FPC-adapterboard, that will make it possible to hook everything up to a breadboard without modifying anything. I will be happy to help with any sort of debugging that might be needed. Just let me know.

Will be documenting the pinout for this LCD also to aid anyone else that might stumble on them. They are very nice

olikraus, 2021-06-16:
Great. It will probably take some more days until I receive the display itself.

olikraus, 2021-07-25:
I recently got access to the display. After wiring with my Arduino it seems to work. U8g2 software development not yet started, I just wanted to see whether the adapter board works.

[image]

spi5498, 2021-07-29:
Hello,

Thank you for your update! This is exciting

I have just traced my pinout for mine. Think i have them all, though, i have for some reason 3 NCs. These is the signal pins i was able to find:

VDD
VSS
E (/WR)
R/W (/RD)
D/I (Not sure how that one is handled)
DB0-7
CS 1-3
RST

Seems like the backlight on this one is permanently hooked to the supply.

Do you have any baseplate code that might give me some initial life from this one?

olikraus, 2021-07-30:
This is the connection table for my current test setup. "D/I" is important and will be controlled by u8g2. Same is true for "E". "R/W" must be connected to GND permanently because U8g2 will only write but never read from the display. "Vee" must be a negative voltage (around -4..-5V). I have just used 4x AA cells to generate the negative voltage.

| Disp. Pin | Disp. Fn. | Arduino Pin | Wire Color  |
|---|---|---|---|
| 1 | GND | GND | black |
| 2 | +5V | +5V | red |
| 3 | Vee | VarPot | - |
| 4 | n.c. | - | - |
| 5 | n.c. | - | - |
| 6 | A (chip select) | A0 | purple |
| 7 | B (chip select) | A1 | purple |
| 8 | C (chip select) | A2 | purple |
| 9 | n.c. | - | - |
| 10 | E (write pulse) | 2 | yellow |
| 11 | R/W (read/write mode) | GND (12) | - |
| 12 | D/I (data/cmd mode) | 3 | green |
| 13 | DB0  | 4 | white |
| 14 | DB1  | 5 | brown |
| 15 | DB2  | 6 | white |
| 16 | DB3  | 7 | brown |
| 17 | DB4  | 8 | white |
| 18 | DB5  | 9 | brown |
| 19 | DB6  | 10 | white |
| 20 | DB7  | 11 | brown |

olikraus, 2021-07-30:
[image]

My own display is actually not a real graphics display, it trippls every 8th row and inserts an empty column every 7th column

olikraus, 2021-07-30:
Another example. U8g2 seems to work so far...

[image]

olikraus, 2021-07-30:
@spi5498 : I have created a new beta release 2.29.9. This release includes the constructor U8G2_T7932_150X32_1 in all of the example. Can you test this with your display? Will something be visible?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

spi5498, 2021-07-31:
Wow, this is pretty exciting! I will try this out during the weekend and report back to you!

I see that my LCD module have a LMC7660 that is probably used to generate the VEE so thats why i couldnt find that when tracing the pinout.

spi5498, 2021-08-01:
Hello!

Did hook up the LCD today.
Unfortunately  no luck with that.

I did confirm the connections once and they were correct from what I can find. Will need to confirm the pinout once more I guess!

not sure if there could be a tweak in the code necessary? There were no life at all

olikraus, 2021-08-02:
Maybe you can share some more details (pictures, wiring, etc) so that I can also crosscheck your setup.

spi5498, 2021-08-02:
I do have it soldered up on a prototype shield for the uno, so not sure how visible that will be.
However, i wonder since i have RST pin on my module, if that needs to be handled in the code?

spi5498, 2021-08-02:
[image]

woho!

After some trial I found that RST needs to be set to high after power on, then it works!

just did it with a jumper wire manually, so need to add this to the code :)

this is very cool! Thank you so much for looking in to this!

spi5498, 2021-08-02:
[image]

spi5498, 2021-08-02:
[image]

spi5498, 2021-08-02:
[image]

olikraus, 2021-08-02:
Wow, looks great. 
Reset pin: You can connect the reset input of your display to any unused GPIO pin. Then pass the number of that GPIO pin as last argument to the u8g2 constructor.

spi5498, 2021-08-02:
Yeah it is really nice! 
Great to see it alive.

Aha! So reset is already implemented in the code for this? Cool!

will also count all the available pixels(width and height) impossible without turning them on.
Haha!

closed 2021-09-02 by olikraus