olikraus/u8g2 #1493: Constructor for EA OLEDS102-6

FrankX0, 2021-05-14:
I am implementing a constructor for an OLED display from EA (OLEDS102-6).
It works fine, except for the first 2 vertical lines: they seem to get overwritten.

This display has a resolution of 102x64.
The internal buffer starts at 13.
In `display_info` is use:
```
  /* tile_width = */ 13,  /* width of 13*8=104 pixel */
  /* tile_hight = */ 8,
  /* default_x_offset = */ 13, 
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 102,
  /* pixel_height = */ 64
```
The issue seems to be in the `tile_width`.
If I set it to 12, the display is not fully updated.
If I set it to 13, the extra bits (104 - 102) seem to wrap-around to the beginning, overwriting the first 2 bits.

How can I solve this?
Thanks in advance for your help.

FrankX0, 2021-05-14:
I think I just found a solution/workaround.
The column address in the SSD1309 is initialised from 13 to 114.
If I add 2 bits to the upper address (116), the wrap-around is gone (they are still being written, but are not mapped to the actual display).

Would this be a valid solution?

olikraus, 2021-05-14:
I am confused. Your display should be a SSD1306.
tile_width should be set to 13, pixel_width to 102, however this will just ensure that the high level procedures write to the visible area. It is the responsibility of your transfer procedure not to exceed those limits.

This could look like this:
https://github.com/olikraus/u8g2/blob/19929687e2f55d1296bf5ba03a0dc73a27947c7a/csrc/u8x8_d_uc1701_dogs102.c#L184-L193

FrankX0, 2021-05-16:
Thanks Oli! I will implement like suggested.

olikraus, 2021-05-24:
thanks for pr https://github.com/olikraus/u8g2/pull/1497 , but it can not be merged directly.
Part of the files are created automatically.

olikraus, 2021-09-02:
According to my research, this seems to be a SSD1306 so I will change the name for it.
Nevertheless, I will probably add this to the SSD1309 file as suggested in the PR

olikraus, 2021-09-02:
The constructor is now called U8G2_SSD1306_102X64_EA_OLEDS102

Thanks for the contribution.

closed 2021-09-02 by olikraus