olikraus/u8g2 #1496: ssd1305 support ZJY223S0700BG03

Sofronio, 2021-05-19:
I used "U8G2_SSD1305_128X32_NONAME_1_4W_SW_SPI" on HellowWorld, and this happened.
[image]
Datasheet is here.
[ZJY223S0700BG03.pdf](https://github.com/olikraus/u8g2/files/6505729/ZJY223S0700BG03.pdf)
This manufactory is called ZJY.

Thank you.

cranefist, 2021-06-26:
Isn't that just mirrored? Have you tried using U8G2_MIRROR ?

olikraus, 2021-06-27:
After u8g2.begin() please insert the command `u8g2.sendF("c", 0xa0);`.
How will the image change with this?

Sofronio, 2021-07-28:
It changed to this. Seems the text was compressed at least by half in height.
[image]

olikraus, 2021-07-29:
Did you also try the SSD1305_128X32_ADAFRUIT constructor?

closed 2022-03-04 by olikraus