olikraus/u8g2 #1498: ABG240128S00-BIW-R  LCD Screen

plazmax, 2021-05-19:
Hello,
I have "ABG240128S00-BIW-R " which has Avant sap1024b driver ic (clone or replica of T6963C) i read too many forum messages and issues page to figure it out but not success yet for correct pining can you help please

Boards: Arduino Mega 2560 , uno, nano
Sketch: Hello World Example 
Libs:  u8g2 and u8glib

constructor:
`U8G2_T6963_240X128_F_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable/wr=*/ 17, /*cs/ce=*/ 14, /*dc=*/ 15, /*reset=*/ 16); // Connect RD with +5V, FS0 and FS1 with GND`

Pin Description:
[image]

[Datasheet 1](http://files.rct.ru/pdf/lcd/abg240128s00-biw-r.pdf)
[Datasheet 2](http://www.artronic.com.pl/pdf/pl/ABG240128S00-BIW-R.pdf)

and i'm confused about "Supply Voltage for LCD" Pin 4 - V0 datasheets show like this:
[image]

Do i need negative supply for it,  or "DD-19.0 | VDD+0.3  " if VDD +5V ; -14 to +5.03V is ok with 20K Pot. and how to connect lcd arduino correctly ?
if im correct for backlight connection like this Pin 1 = A = +5V, Pin 2 = K GND

olikraus, 2021-05-19:
> Do i need negative supply for it, or "DD-19.0 | VDD+0.3 " if VDD +5V ; -14 to +5.03V is ok with 20K Pot. and how to connect lcd arduino correctly ?

There seems to be a negative voltage converter in the PCB, so you need to follow figure 1 in datasheet 2

> if im correct for backlight connection like this Pin 1 = A = +5V, Pin 2 = K GND
It looks like that you need a resistor here. Maybe a 33 Ohm type, but you could also start with 100 Ohm and lower it a little bit. At least the current should not exceed 180mA

Rest looks good (of course I can not validate your wiring)

plazmax, 2021-05-20:
Thank you for quick reply, here my wiring but 2 of them i cant figure connect where or NC what do you think am i correct about wiring ?

FUNCTION | PIN | LCD | ARDUINO
-- | -- | -- | --
Power supply for backlight(Currect:150mA;Reference voltage:5.0V) | 1 | A | +5V
Power supply for backlight(-) | 2 | K | GND
Output of supply negative voltage by the DC-DC converter on the module | 3 | VEE | POT LEG 1
Power supply for the LCD drive | 4 | V0 | POT LEG 2
Control pin for oscillation（H：normal operation，L：shutdown | 5 | PD | ???
Powerr ground (0V) | 6 | GND | GND
Power supply for Logic(+5.0V) | 7 | VDD | +5V
LCD output voltage | 8 | V5 | ???
Write Signal | 9 | /WR | 17
Read Signal | 10 | /RD | +5V
Chip Selection Signal | 11 | /CE | 14
Data or Instruction Select Signal (H: data register L: instruction register) | 12 | C/D | 15
Reset Signal | 13 | /RST | 16
Data Bus Line | 14 | DB0 | 8
Data Bus Line | 15 | DB1 | 9
Data Bus Line | 16 | DB2 | 10
Data Bus Line | 17 | DB3 | 11
Data Bus Line | 18 | DB4 | 4
Data Bus Line | 19 | DB5 | 5
Data Bus Line | 20 | DB6 | 6
Data Bus Line | 21 | DB7 | 7
Font Selection Terminal | 22 | FS | GND

Here pics if someone need them i'm adding too
[image]
[image]

You are correct there is negative voltage converter AI1652CS or GS and LM224

olikraus, 2021-05-21:
I never saw something like V5 and PD, but after reading the datasheet, I would suggest:
V5 --> no connection
PD --> 5V

olikraus, 2021-05-21:
Just one more point regarding the backlight display:
Do not connect the A to Arduino +5V directly, because
a) Arduino is not able to provide 150mA (it can max provide 100mA) and will probably go into reset state.
b) You will probably destroy the backlight if there is no resistor between LED A and the power source. 
As mentioned above, there should be 33 Ohm resistor in between somehow. More over, the 33 Ohm resistor must be a 1 Watt type. A usual 1/4 or 1/8 Watt type will get rosted soon.

Anyhow you could test the complete setup without Backlight first. Everything will be visible without backlight, so leave LCD A and LCD K unconnected until u8g2 works correctly.

olikraus, 2021-05-21:
Here is my own setup:
[image]

The backlight is derived from a 9V battery. The big green resistor is placed on the left side of the display (lower left in the picture).

plazmax, 2021-05-23:
I updated the table in line with the information you provided and after trying it, I will inform you, thank you very much.

closed 2021-12-27 by plazmax

mcv-m, 2022-11-05:
Hello, 
what you have written here is very helpful to me. It also has a GLCD with 22 pin and A' Leonardo. Can I choose pins (Leonardo=/LCD)
A0=/WR, 
A1=/RD, 
A3=/CE, 
A4=/CD, 
A5=/RST
I'm just starting to gain knowledge in the Arduino and LCD environment. All help is priceless for me. I've the same LCD as my colleague "plazmax"
Many greetings.

olikraus, 2022-11-05:
> Can I choose pins 

yes, but then you need to adjust the constructor accordingly:

```
U8G2_T6963_240X128_F_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable/wr=*/ A0, /*cs/ce=*/ A3, /*dc=*/ A4, /*reset=*/ A5); // Connect RD with +5V, FS0 and FS1 with GND
```
A1 is not required, instead /RD should be connected to +5V

mcv-m, 2022-11-06:
Thanks to you @olikraus  and @plazmax  for helping me in my first steps with Arduino and GLCD. I attach a few pictures from the operation of the LCD and the solutions with which my LCD works! :)
[image]
[image]
[image]

olikraus, 2022-11-06:
Looks good 😊