olikraus/u8g2 #1501: UC1601 128x64 pixels help

folny, 2021-05-23:
Hi

I'm trying to use u8g2 and UC1601 128x64 display on my stm32f4 but it still doesn't work for me I tried to use the settings for ST7565 but still nothing, does anyone have any experience with UC1601 128x64 Display ?.

folny, 2021-05-24:
Which parts of the library need to be modified to reach 128x64 pixels ? Are changes to the u8x8_d_uc1601.c file not enough? or is it harder to do it ?.
 
#include "u8x8.h"

static const uint8_t u8x8_d_uc1601_128x32_powersave0_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x0af),		                /* display on */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

static const uint8_t u8x8_d_uc1601_128x32_powersave1_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x0ae),		                /* display off */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

static const uint8_t u8x8_d_uc1601_128x32_flip0_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x0c4),				/* bit 1: MX, bit 2: MY */
  U8X8_C(0x060),		                /* set display start line to 32 */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

static const uint8_t u8x8_d_uc1601_128x32_flip1_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_C(0x0c2),				/* bit 1: MX, bit 2: MY */
  U8X8_C(0x040),		                /* set display start line to 0 */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

static const u8x8_display_info_t u8x8_uc1601_128x32_display_info =
{
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
  
  /* post_chip_enable_wait_ns = */ 1,	/* uc1601 datasheet, page 46 */
  /* pre_chip_disable_wait_ns = */ 5,	/* uc1601 datasheet, page 46 */
  /* reset_pulse_width_ms = */ 1, 
  /* post_reset_wait_ms = */ 6, 
  /* sda_setup_time_ns = */ 12,		/* uc1601 datasheet, page 44 */
  /* sck_pulse_width_ns = */ 15,	/* uc1601 datasheet, page 44 */
  /* sck_clock_hz = */ 2000000UL,	/* */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 1,
  /* data_setup_time_ns = */ 60,	/* uc1601 datasheet, page 43 */
  /* write_pulse_width_ns = */ 80,	/* uc1601 datasheet, page 43 */
  /* tile_width = */ 16,		
  /* tile_hight = */ 4,
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 4,
  /* pixel_width = */ 128,
  /* pixel_height = */ 32
};

static const uint8_t u8x8_d_uc1601_128x32_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  U8X8_C(0x0e2),            			/* soft reset */
  U8X8_C(0x0eb),            			/* LCD Bias: 0xe8: 6, 0xe9: 7, 0xea: 8, 0xeb: 9 */
  //U8X8_C(0x023),            			/* 0x020...0x023 only for UC1601, not for UC1601s */
	

  //U8X8_C(0x02e),            			/* LCD Load + Internal Charge Pump (default: 0x2e) */
  U8X8_C(0x024),		                /* Temperature Compenstation, default: 0x24 */
  U8X8_C(0x089),		                /* RAM address ctrl, default: 0x89 */
  U8X8_C(0x0c4),		                /* RAM mapping ctrl */
  U8X8_C(0x0a0),		                /* Frame Rate, 0x0a0 or 0x0a1 */
  U8X8_CA(0x081, 0x0df),		/* set contrast */
  U8X8_C(0x02e),            			/* LCD Load + Internal Charge Pump (default: 0x2e) */	
  U8X8_C(0x060),		                /* set display start line to 32 */
  
  U8X8_C(0x0a6),		                /* normal display */
   
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

uint8_t u8x8_d_uc1601_128x32(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t x, c;
  uint8_t *ptr;
  switch(msg)
  {
    case U8X8_MSG_DISPLAY_SETUP_MEMORY:
      u8x8_d_helper_display_setup_memory(u8x8, &u8x8_uc1601_128x32_display_info);
      break;
    case U8X8_MSG_DISPLAY_INIT:
      u8x8_d_helper_display_init(u8x8);
      u8x8_cad_SendSequence(u8x8, u8x8_d_uc1601_128x32_init_seq);
      break;
    case U8X8_MSG_DISPLAY_SET_POWER_SAVE:
      if ( arg_int == 0 )
	u8x8_cad_SendSequence(u8x8, u8x8_d_uc1601_128x32_powersave0_seq);
      else
	u8x8_cad_SendSequence(u8x8, u8x8_d_uc1601_128x32_powersave1_seq);
      break;
    case U8X8_MSG_DISPLAY_SET_FLIP_MODE:
      if ( arg_int == 0 )
      {
	u8x8_cad_SendSequence(u8x8, u8x8_d_uc1601_128x32_flip0_seq);
	u8x8->x_offset = u8x8->display_info->default_x_offset;
      }
      else
      {
	u8x8_cad_SendSequence(u8x8, u8x8_d_uc1601_128x32_flip1_seq);
	u8x8->x_offset = u8x8->display_info->flipmode_x_offset;
      }	
      break;
#ifdef U8X8_WITH_SET_CONTRAST
    case U8X8_MSG_DISPLAY_SET_CONTRAST:
      u8x8_cad_StartTransfer(u8x8);
      u8x8_cad_SendCmd(u8x8, 0x081 );
      u8x8_cad_SendArg(u8x8, arg_int );	/* uc1601 has range from 0 to 255 */
      u8x8_cad_EndTransfer(u8x8);
      break;
#endif
    case U8X8_MSG_DISPLAY_DRAW_TILE:
      u8x8_cad_StartTransfer(u8x8);
    
      x = ((u8x8_tile_t *)arg_ptr)->x_pos;
      x *= 8;
      x += u8x8->x_offset;
      u8x8_cad_SendCmd(u8x8, 0x010 | (x>>4) );
      u8x8_cad_SendCmd(u8x8, 0x000 | ((x&15)));
      u8x8_cad_SendCmd(u8x8, 0x0b0 | (((u8x8_tile_t *)arg_ptr)->y_pos));
    
      c = ((u8x8_tile_t *)arg_ptr)->cnt;
      c *= 8;
      ptr = ((u8x8_tile_t *)arg_ptr)->tile_ptr;
      do
      {
	u8x8_cad_SendData(u8x8, c, ptr);	/* note: SendData can not handle more than 255 bytes */
	arg_int--;
      } while( arg_int > 0 );
      
      u8x8_cad_EndTransfer(u8x8);
      break;
    default:
      return 0;
  }
  return 1;
}

olikraus, 2021-05-24:
> I'm trying to use u8g2 and UC1601 128x64 display on my stm32f4 but it still doesn't work for me I tried to use the settings for ST7565 but still nothing, does anyone have any experience with UC1601 128x64 Display ?.

There should be something visible with the existing constructors.

Do you use an Arduino environment? In most cases the wiring is wrong or you applied the wrong values.
You should first start to describe your setup in much more detail.

folny, 2021-05-24:
I'm using Keil uVision my connection is correct the display works well but I can only see half of the 128x32 lines on it.

it is this display https://www.tme.eu/Document/59369f5729330dfc5cc7350507d991b2/DEM128064QFGH-PW-dte.pdf

olikraus, 2021-05-24:
ok, understand. I have created the 128x64 constructor in beta release 2.29.5, please check...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

folny, 2021-05-24:
Thank you for updating your library, the display works almost well, but you still have to change the line U8X8_C (0x060) in the file u8x8_d_uc1601.c, / * set display start line to 32 * / to U8X8_C (0x040), / * set display start line to 32 * / otherwise the display starts typing from the popovice. Another problem is that it is shifted about 4 pixels to the left as seen in the photo here is a test code which should render a frame of 128x64 pixels.

  u8g2_Setup_uc1601_128x64_f(&u8g2,U8G2_R0,u8x8_byte_stm32_hw_spi,u8x8_stm32_gpio_and_delay);  
  u8g2_InitDisplay(&u8g2);
  u8g2_SetPowerSave(&u8g2,0);	
  u8g2_SetContrast(&u8g2,150);
  u8g2_SetDrawColor(&u8g2,1);
  u8g2_DrawFrame(&u8g2,0,0,128,64);
  u8g2_SendBuffer(&u8g2);
  u8g2_ClearBuffer(&u8g2);

[image]

folny, 2021-05-24:
I solved the problem by changing / * default_x_offset = * / 2, so now everything works as it should. Thank you again for your help.

olikraus, 2021-09-02:
After a long delay, I have updated the code. Thanks for the feedback.

closed 2021-09-02 by olikraus